nielsonm236/NetMod-ServerApp #124: 20221109 0109 Pre-Release for QOS 1  possible problem using code uploader

TheArtizan, 2022-11-18:
when using the the code uploader with 20221109 0109 Pre-Release for QOS 1 testing, some times I get the following-       a:,'>    ine the top left of my browser.
[image] I have managed to get it to upload after coupling the SWIM an reloading. I have tried this on three seperate modules one worked first time ok but others had this problem. Thank you for this great work.

nielsonm236, 2022-11-19:
A few questions just to make sure:
- You have the external EEPROM? (must be ... or you wouldn't get Uploader to respond at all).
- When you did the upload did you remember to upload the following in this order: 1) The new Uploader (seems redundant since the old Uploader is running, but the new Uploader needs to be loaded); 2) The new Strings file; 3) The new code.

As FYI, I haven't seen the issue described here with the current pre-release code, but I do recall having trouble because I didn't follow my own procedure (above). :-)

Also, see issue #122. I'm likely to abandon the QOS 1 effort because the MQTT spec requires a disconnect/reconnect for dropped messages to be resent. Makes the QOS 1 approach rather useless. Instead I'm back to digging into the QOS 0 problems when there are a batch of PUBLISH state changes from HA.

TheArtizan, 2022-11-20:
Thanks for your reply, I have the external as ram specified, I did do as you sugested and install the new uploader then the new strings then the new code. I did note that the web relays that I recieve from China during October 2022 have the chip SM8S0105C and not the STM8S005C6 you have, so I did change the Device on the ST Visual programmer to SM8S0105C via the SWIM USB. I was then able to use the old code uploader to install the 20221109 0109 Pre-Release for QOS 1. I think that the reason I got this is maybe because of the device missmatch when coupled to the SWIM and ST Visual Programmer. The devices seem to work as you intended so far but I will keep testing using Home Assistant, Thank you again from the UK

nielsonm236, 2022-11-20:
Given you followed the procedure the symptom is very puzzling. Were these devices in a test or production setup where something might have been generating a lot of MQTT requests while you upgraded (and I mean a LOT of MQTT requests)? Given the bug (see #113) I'm trying to track down right now that might explain it ... otherwise I'm at a loss.

TheArtizan, 2022-11-20:
I was testing these for use in my home , there were no excessive MQTT requests that I am aware of. It happened only while tring to install via the new uploader, but all seems ok after I plugged back in with the SWIM, then reinsalled the lastest working uploader and changed the device to STM8S105 (to match the chip on my board). So maybe the problem could be me uploding with the wrong device STM8S005C. Thank you so much, I will send some pictures of a neat and robust way to connect the off board RAM module, all the best

nielsonm236, 2022-11-20:
Just so I understand (and in case there are notes I need to add to the manual):
When you say you "changed the device to STM8S105" do you mean just when you used the SWIM programming interface?
Or are you doing your own recompiles of the code?
I want to be sure the releases I do can accommodate both device types as users can't specify which chip they will get.
Mike

TheArtizan, 2022-11-20:
I only changed the device to the STM8S105 when I used the SWIM interface. I am not doing my own code. Many thanks

nielsonm236, 2022-12-17:
Addressed in release 20221217 1925 via addition to the manual.

closed 2022-12-17 by nielsonm236