nielsonm236/NetMod-ServerApp #128: Look into "Configuration Save", "Reboot", and "Boot State" processes again

nielsonm236, 2022-12-02:
This might be a bad idea so I'm looking for some feedback.

Long ago we implemented the concept of "Boot State" so that we could be assured that the Output pins were in a known state at power on. This would put each Output pin in an ON or OFF  state with a setting for each pin. I think that is a good idea and needs to remain the way it is for power-on.

However, during test where I'm making minor changes to Configuration and booting many dozens of times across 5 to 10 modules it has started to occur to me that Boot State might be an annoyance when there wasn't really any need to change the state of the Output pins to their Boot State.

As an example let's say I have 8 Outputs defined and they are all in some ON / OFF state, then I define a 9th Output that was previously Disabled. I click Save, and the module reboots to save the new Configuration. The ON/OFF state of my previously defined 8 Outputs go to their Boot State as part of the reboot. Well, this is fine in test, but what if in the real world one of those outputs controlled a garage door, a gate, or lighting in a critical area? We might not want that Output to go to Boot State if it didn't absolutely need to only because I made a minor change to Configuration.

It occurs to me that I might be able to determine that we are "restarting" as opposed to "rebooting" when a Save is clicked. In that way I could probably leave the existing Outputs in their prior state because there was no power loss. Of course, if the user actually clicked on "Reboot" I would treat it as a power loss reboot and take all outputs to their Boot State. But if the user clicked on Save I could filter for changes that actually require taking all Outputs to their Boot State rather than ALWAYS taking all Outputs to their Boot State.

Well, that sounds OK to me, but there might be some downsides:
1) Right now clicking Save on the Configuration page ALWAYS takes all Outputs to their Boot State. You know it will happen. If I make the change then a Save usually won't take all outputs to their Boot State. But sometimes it will. This might be confusing.
2) This change might leave behind an Output setting that doesn't get cleared. For instance, I have an Output that happens to be ON, then I change the pin to Disabled. Later I change that pin back to being an Output. If no Reboot occurs between those changes the module likely will put that Output in the ON state.
3) Similar to (2), a pin is changed from an Input (which is in the ON state) to an Output. The Output pin might then be in the ON state after Save is clicked.
4) Configuration changes that I think MUST require a Reboot on clicking Save, thus putting all Outputs in their Boot State:
- Changing any of these settings: Network Module Name, IP Address, Gateway, Netmask, Port, MAC Address.
5) Configuration changes that I think might NOT require a full Reboot:
- Clicking Full Duplex mode feature. Requires re-running the ENC28J60 initialization, which can cause loss off in-flight TCP packets. Of course, a full Reboot also does that. See Note 1.
- Clicking HA Auto feature. Changing this requires re-running the MQTT startup code so this needs careful review. See Note 1.
- Clicking MQTT feature. Changing this requires re-running the MQTT startup code so this needs careful review. See Note 1.
- Changing any of these settings: MQTT Server IP, MQTT Port, MQTT Username, MQTT Password. Changing these requires an MQTT Disconnect and re-running the MQTT startup code so this needs careful review. See Note 1.
- Clicking the Disable Cfg Button feature
- Clicking DS18B20 feature. This requires re-running the HA Config message generation code.
- Changing pin Type.  This requires re-running the HA Config message generation code.
- Clicking Invert setting
- Changing Boot State
- Changing IO Name. This is in the Browser Only Config, I think it already just does a Restart.
- Changing IO Timers. This is in the Browser Only Config, I think it already just does a Restart.
6) If something in (5) above ends up requiring a lot of code I might have to revert that item to "do a full reboot if changed" like it is now. That's where I worry that a user might find this to be non-intuitive. Of course I'll write a page in the Manual on this ... but it's better if it is intuitive.

Note 1: Actions with "See Note 1" might also cause loss of inflight incoming MQTT TCP packets, which might cause receipt of a malformed MQTT message, which might in turn cause a Reboot. If a Reboot occurs all outputs will go to their Boot State. I think this would only occur if there was MQTT automation running that is generating state changes to the Network Module at the time that the Configuration Save occurred.

I don't think it is much code work to implement this idea. A new REST command (perhaps /92 for Restart whereas /91 already exists for Reboot), and a small change in Javascript to call /92 on Save instead of /91. The concept of Restart vs Reboot already exists in the main.c code, but the selection of when to run Restart vs Reboot would need to be more granular. Test will be very difficult.

I will leave this in "think about it" mode for a while.

yozik04, 2022-12-04:
The cleanest would be always reset state to initial on reboot/restart. That is the simplest code. No need to overcomplicate things. But it is your hobby and your decision =)

nielsonm236, 2022-12-04:
No one complained ... just me noticing this might be a problem as a result of doing hundreds of reboots during testing. I'll leave it alone then. :-)
I do agree the idea here is much more complicated, and therefore more likely to have a bug. Don't need any more of those!

closed 2022-12-17 by nielsonm236