nielsonm236/NetMod-ServerApp #133: STM8S105 Device, can't figure it out from documentation

amiiit, 2022-12-26:
Hello and thank you for the promising tool!

I have a STM8S105 device (photo attached), but I wasn't able to flash it:

I am trying over the following 2 setups using a blue ST-LINK V2 USB stick (STM8 & STM32):

1. Intel Mac running VirtualBox with Windows 10 and the recommended software from the documentation
2. Raspberry PI 3+ with [texane/stlink](http://github.com/texane/stlink) installed on it, controlling it over SSH an attaching the USB programmer directly to the RPi board.

Sorry, I don't have a native Windows installation but surely people got it working on mac or Raspberry PI? If it's only about the windows installation I would definitely set one up, _but few other things aren't clear to me:_

What confuses me is the wiring in the documentation, which isn't absolutely matching to the diagram of the USB Stick (Compare page 55 with page 60, screenshots from the manual attached)

### My USB Programmer has the same pinout as the one from the documentation
[image]

### The actual wiring example shows a different pinout, perhaps used a different version of the programmer?
[image]

**Is there a way to figure out the programming pin order from observing the device itself? Perhaps they come in different order now?**

### What Device setting to choose for the STMS105?
One more things that's unclear to me, is which of the 2 options, **STMS105x4** or **x6** do I need to select? I can't get both of them to work and I don't know if that's this setting, the wiring, or perhaps my funny VirtualBox setup.
[image]

## Here is a photo of the device I am using:
[image]

## And my setup
Driving 5v to the device from the RPi GND/5V pins:
[image]

Thank you so much for the help 🙏

nielsonm236, 2022-12-26:
Hi @amiiit 
Great photos. Answered several questions right away. The markings on the chip say it is a 105C6T6, so the 105x6 selection in the Device menu should be correct. I don't have any of the 105 devices, but I've heard from a couple of other people that they programmed OK for them - from this I think the underlying board has not changed. So we've just got to figure out what the missing part is for you.
The ST-Link to HW-584 wiring looks correct. I assume the wires go to the top row of the ST-Link dongle. Looks like it in the photo but not sure.
I assume the ST Visual Programmer did not give you a "cannot communicate with the tool" message. That would indicate if the Programmer can see the ST-Link device.
If all that looks OK then the next thing is to make sure you cleared the ROP bit. Find this part in the manual:
[image]

And now I'm short on ideas other than making sure the Dupont wires all have good connections.

Mike

nielsonm236, 2022-12-26:
AHHHHH ... Just looked closer at your comments and the photos. Your ST-LINK is different than mine. The GND and SWIM pins are reversed. Here's a photo of mine. I will send a photo in a few minutes.

nielsonm236, 2022-12-26:
Yours:
[image]

Mine:
[image]

Now, why in the world would they do that? I wonder if it is a ST-Link knock-off issue. I will do a little research.
Mike

jmcvieira1, 2022-12-26:
Hi Mike
 @nielsonm236 

In the Photos the red led at the network module isn´t so posed to bee on?
I have a red ST-Link  like your´s but i have a blue in the box never open never used and can confirm that the GND and SWIM pins are reversed at least in the marking´s i will test´it and post the result.

nielsonm236, 2022-12-26:
Looks like this has been a problem going back 2 or 3 years. You are the lucky winner of the "different" ST-Link device!
See https://forums.adafruit.com/viewtopic.php?f=19&t=139902

With a little more searching I find pinout diagrams showing either pinout. But each resource only talks about one or the other pinout - NO mention that there are two pinouts out there.

I will have to update the manual so that people are aware they could get either one.
Mike

nielsonm236, 2022-12-26:
On mine the ST-Link LED turns on as soon as I plug it into the USB port. But that LED is also on if I have the ST-Link wired to my HW-584 but NOT plugged into the USB port. So I believe the LED is just a "power" LED, and it can get power from either the USB port OR the HW-584 board. The LED on my ST-Link device is blue.

The LED on the HW-584: I don't recall clearly but I think with the original manufacturer FW the LED is OFF when powered up. With my FW the LED will be ON, and I have various blink sequences depending on reboot source, whether the user presses the Reset button, etc. Most of those blink sequences are of no use except for developers.

Mike

jmcvieira1, 2022-12-26:
Mike
i tested the blue one and it worked fine following the marking´s on the pen.
I pay attention to the led and it performs as you described on a already flashed board.

nielsonm236, 2022-12-26:
jmcvieira1 - You had a ST-Link with markings like @amiiit ?

jmcvieira1, 2022-12-26:
Yes same color and same markings like @amiiit

amiiit, 2022-12-26:
Hello @nielsonm236 @jmcvieira1 ,

Pleasure to meet you and read your kind and helpful replies! 

> But that LED is also on if I have the ST-Link wired to my HW-584 but NOT plugged into the USB port. So I believe the LED is just a "power" LED, and it can get power from either the USB port OR the HW-584 board. The LED on my ST-Link device is blue.

I observed that the red LED on my ST-Link stick **is flashing**. The USB plug is disconnected, and it's wired to the HW-584, which receives 5v from the RPi. When the stick is connected to the computer's USB port, the light is always on (red).

> I assume the ST Visual Programmer did not give you a "cannot communicate with the tool" message. That would indicate if the Programmer can see the ST-Link device.

Yes, I do get this error: _Cannot communicate with the tool_, when running STPV on a win 10, VirtualBox. 

> The ST-Link to HW-584 wiring looks correct. I assume the wires go to the top row of the ST-Link dongle. Looks like it in the photo but not sure.

Yes, top row! I understand the schema as if one would look at the port from the above.

So unless that flashing LED described is any indication helpful to something, I could then only assume that it lies on the VirtualBox? 

Thanks, Amit.

jmcvieira1, 2022-12-26:
Hello Amit
I also work with win 10, let´s see if i can help you a bit, from the error that you mention the STPV is not finding the usb ST-link dongle.
do you have the ST-Link dongle properly installed on win 10, after you insert the dongle on a usb port can you see it´s name if you clik on the Safely Remove Hardware Icon on the system tray?
[image]

It should look like mine: 
[image]

amiiit, 2022-12-27:
✨ This is awesome! 🚀 

I was just able to get this working with the VirtualBox!! Thank you so much for your guidance that helped me narrowing down on the issue. In fact, pointing me to the indicator in Window's bottom bar cracked this for me, such a trivial error. **_The USB stick simply wasn't correctly mounted by the system_** 🙈 😇 🤦 

## Here are the steps I needed to make it work via VirtualBox on an Intel Mac (MBP Late 2011)

**First, download Windows 10 image from the above resource:**
https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

* Selected MSEdge on Win10 (x64)
* Selected 'VirtualBox' as VM Platform.
* Downloaded the image and imported it into Oracle's VirtualBox

#### The crucial part, USB configuration:
Besides all the other configurations, like giving the VM enough memory, CPU power etc', we need to allow-list the USB device we're going to connect, which in this case is the ST-Link USB Programmer.

Under _Settings -> Ports -> USB -> USB Device Filters_ click _Add Filter_ and choose _STMicroelectronics STM32 STLink_ (or similar).

After adding the USB Filter the windows machine was able to detect the USB Programmer and use it without any errors. For the installation and usage of the software, including the usage of the `STM8S105x6` I simply followed the manual.

See attached screenshot for the additional configuration on VirtualBox:

[image]

I am trying now to do the same on an M1 Mac and I'll report if the same process worked for me.

amiiit, 2022-12-27:
I can now finally connect this to my irrigation system and control the water for my trees and garden! 🌳 🥑 🍅 🥬 🍓 

My previous attempt with the WebRelay v1 was not robust enough to work, and it's been over 6 months since, so I'm quite excited to try this with a robust firmware and using MQTT 😍 

Thank you @nielsonm236 @jmcvieira1 I hope I can find a way to contribute to this project, with a tutorial, feel free to approach me on any browser, frontend, JavaScript issues as this is my daily bread.

amiiit, 2022-12-27:
For completeness. I tried to run the windows 10 image on VirtualBox on a Mac M1. The image could not start and the whole thing crashes. It was not possible to run windows on M1 VirtualBox. Once that is fixed (by Oracle? Microsoft?), I will try again and report.

nielsonm236, 2022-12-27:
Good this was figured out. Thank you @jmcvieira1 for your help.
@amiiit Send me an email at nielsonm.projects@gmail.com some time so I can reach you for code contributions. It is always good to have another GUI person to ask questions or make suggestions. I will always be a novice at everything. :-)

closed 2022-12-27 by nielsonm236

jmcvieira1, 2022-12-27:
@amiiit 
I´m glad you figured out, i´m here to help.
Good work!