nielsonm236/NetMod-ServerApp #135: After flash, several errors

amiiit, 2022-12-29:
Hello,

Thank you again for the help in in previous issue, and so I was able to flash and then connect to a router that allows the default IP to be assigned to the device, great!

I have made the following observations:

Calling the endpoint `http://192.168.1.4:8080`  returns the following:

```
% curl -i http://192.168.1.4:8080
HTTP/1.1 200 OK
Content-Length:00567
Cache-Control: no-cache, no-store
Content-Type: text/html; charset=utf-8
Connection:close

���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������%
```

Assuming this is binary, I dropped the output into a file:

```
curl http://192.168.1.4:8080 > file.bin
```

And using `less` to read it, I can see a bunch of `<FF>`, I didn't count them but `ls -l` shows the file has length of 567 (bytes?)

```
<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF<FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF><FF>
```

I'm getting  similar output also for `/61` and `/60`

Further observations:

The endpoint `http://192.168.1.4:8080/98` consistently returns `1001111111111111`, despite my attempts to flip the switches:

```
% curl http://192.168.1.4:8080/98
1001111111111111%
% curl http://192.168.1.4:8080/00
% curl http://192.168.1.4:8080/98
1001111111111111%
% curl http://192.168.1.4:8080/01
% curl http://192.168.1.4:8080/98
1001111111111111%
% curl http://192.168.1.4:8080/55
% curl http://192.168.1.4:8080/98
1001111111111111%
% curl http://192.168.1.4:8080/56
% curl http://192.168.1.4:8080/98
1001111111111111%
```

The server always answers with 200, so I assumed the operation was successful, but it seems like something is off.

Network Statistics Page seem to load correctly and gives the following output:

. | .
-- | --
0000000002 | Dropped packets at the IP layer
0000000017 | Received packets at the IP layer
0000000011 | Sent packets at the IP layer
0000000000 | Packets dropped due to wrong IP version or header length
0000000000 | Packets dropped due to wrong IP length, high byte
0000000000 | Packets dropped due to wrong IP length, low byte
0000000000 | Packets dropped since they were IP fragments
0000000000 | Packets dropped due to IP checksum errors
0000000000 | Packets dropped since they were not ICMP or TCP
0000000000 | Dropped ICMP packets
0000000000 | Received ICMP packets
0000000000 | Sent ICMP packets
0000000000 | ICMP packets with a wrong type
0000000000 | Dropped TCP segments
0000000018 | Received TCP segments
0000000014 | Sent TCP segments
0000000000 | TCP segments with a bad checksum
0000000000 | TCP segments with a bad ACK number
0000000003 | Received TCP RST (reset) segments
0000000000 | Retransmitted TCP segments
0000000000 | Dropped SYNs due to too few connections avaliable
0000000000 | SYNs for closed ports, triggering a RST

Initially I thought this may have to do with the build type, I indeed had an upgradeable browser only build, but I did not do the hardware upgrade. So I reflashed with the non-upgradeable build of the browser-only version, and it seems like it's similar behaviour. 

After reading more issues here I started to suspect the power supply, that it got from the RPi. I replaced it with a high-quality USB supply which gives a constant 5.2V 

The device has nothing else connected to it, and it was successfully flashed with the latest Browser-Only version just now.

*Any idea what I may be doing wrong?* 

Cheers,

Amit

nielsonm236, 2022-12-29:
Yes - the initial problem was use of the upgradeable version without the external EEPROM.
I'm away from home right now but will try to replicate what you are seeing when I get home in a few hours.
Mike

nielsonm236, 2022-12-29:
Let's ignore the parts where you loaded the upgradeable file.
I just re-tested the non-upgradeable build and it worked OK here. The file I loaded:
NetworkModule-Browser.sx from the 20221217 General Release. I programmed it with the STVP programmer (over the SWIM interface).
Note that you should program using the "Program / Address Range" method per this snippet from the manual, but if you forgot to do that all that will happen is you will lose any  IO Names and IO Timer values. But you WILL get a programming error message and you must acknowledge that message to complete the process. Then the module should respond as expected.
[image]

So ... I believe the .sx file on GitHub is OK, but we still need to understand why it isn't working for you. So let me ask a few more questions to see if one of these provokes an idea:
a) I noticed you kept the original 192.168.1.4 IP address. Nothing wrong with that as long as you are sure there isn't a conflict on your network.
b) Do you have more than one HW-584 device on the network? If yes, are you sure you gave each one a unique MAC address?
c) I still suspect you may not have programmed with the NetworkModule-Browser.sx file. Is that possible? Since you are getting some responses from the HW-584 module it looks like one of the .sx files did get programmed ... I just can't tell which one.

Mike

nielsonm236, 2022-12-29:
Actually my (a) comment is incorrect as you didn't get to the Configuration page yet. So (a) should really just be about whether you have any conflicts on the Network. Or perhaps you were still direct connect to the module? Not on a network yet?

nielsonm236, 2022-12-30:
Also - let me know what processor your board came with. I'm finding some people have received the STM8S105C6 chip rather than the STM8S005C6. I don't have a board with the 105 chip so I don't know if this could be causing problems for you, but worth checking. I'm reviewing the manual again with regard to this issue and may have to make some updates. If you are using the .stp file it calls out the STM8S005C6 - you should edit the .stp file and change the DEVICE setting to the 105 (if that is what you have).

amiiit, 2022-12-30:
Okay, first thing off, I was able now to correctly flash the browser only version, configure and change the device's IP and now it's running happily in my local network!! I will continue to the next step of connecting it to the relay board, where once the WebRelay v1 was connected 🤞 

Now to the technical issue, the moment when it actually succeeded was when STVP failed to verify the memory 🤷 . I think I understand it now, that's because it starts with the configuration on the memory, so that of course it can't be the same like the image, still it puzzles me when I had a success before and it kept the previous version on the device, now I get failure and that is actually when the device flashed successfully?

If you want, I can make further experiments? I bought a handful of these so I'm not afraid to flash them as much as required.

So yea, thanks so much, I think I will now solder the connector to the relay board and read the manual parts regarding this connection. When this works for me, I intend to switch to the MQTT version and starting pouring some logic into it.

Please let me know if I could provide any helpful information, yes I have the 105 chip, attaching foto:

[image]

Have a great day/night, and a happy new year!

nielsonm236, 2022-12-30:
This is great news!!! 
Yes, there are two kinds of errors you'll see during programming.
1) Messages that say something like "... address 0xD is out of range and is ignored ...". This is normal. This just has to do with the way the compiler/linker creates the binary file and specifies some memory areas that are not in the programming range.
2) I can't remember the error message, but if you are programming the Browser Only versions and forget to use the “Program / Current Tab”, “Program / Address Range”, "8000 to FEBF" method you get a popup indicating a mis-compare error ... which means you over-wrote the Flash storage used for the IO Names and IO Timers. If you acknowledge by clicking "OK" the device will still boot, but you'll have to re-enter your IO Names and IO Timers information in the GUI. That's a pain, but not fatal.

So I'm not sure what got missed, but somehow you weren't getting it programmed correctly. Now you are - so you're on your way! I opened a new issue (#136) to clarify the programming steps. While looking at your issue I found that part of the Manual has become more confusing over time and needs to be cleaned up.

closed 2023-01-04 by amiiit