nielsonm236/NetMod-ServerApp #141: Selectable pinout branch

peoyli, 2023-02-13:
When I have the 'standard 16 channel relay' pin order selected (/822):
```
	// HW-584 silkscreen
	// 5V 16 15 14 13 12 11 10 09 G
	// 5V 08 07 06 05 04 03 02 01 G
	// IO Locations
	// 5V 02 04 06 08 10 12 14 16 G
	// 5V 01 03 05 07 09 11 13 15 G

```
I am unable change status of relay 2 (iopin 16 on silkscreen). I do not have DS18B20 activated (only use is to control relays).

When I shift the controller (lower pin row, 08-01 connected to even row of relay control pins 02-16), I can confirm that relay 2 works as expected.
I have also verified each of the relays by connecting their control pin to ground.

In 'normal' (/821) pin mode, all relays are controllable.

nielsonm236, 2023-02-14:
I've reviewed the code and don't see how this is possible. So I will need to do some debug to track it down. 
Since I haven't released this feature yet you must have done a build from my "in progress" code, is that correct? 
Is your test based on a Browser Only build or a MQTT build? Knowing this will help me replicate what you are seeing.

Just FYI the feature I plan to release in parallel with the Selectable Pinout code is the "PCF8574" code. I am starting debug on the PCF8574 code this afternoon if I can. PCF8574 has a very large number of code changes, so it could take some time to debug. If I get into trouble with that, and I am not finding the problem you describe, I could email a test build to you to try.

Mike

peoyli, 2023-02-14:
Your assumptions are correct: I've built it myself from the branch with the code for selectable pin order, 'BUILD_TYPE_BROWSER_STANDARD' (line 277), and in that configuration block also changed line 321 "PINOUT_OPTION_SUPPORT" to activate the pin order configuration.
Both changes in uipopt.h
The bug should be easy to reconstruct, but as you mention, the code seems fine (pin mapping tables).
Could there be code in other places of this branch that disables actions on physical pin 16 if any custom pin order has been selected ?

nielsonm236, 2023-02-14:
I just ran a test here using an MQTT build, but with the MQTT feature boxes unchecked (actually, all configuration boxes unchecked). It worked as expected with /821 and /822. I'm using an LED display board to show me the state of all output pins. I also tried with "Invert" check for all pins, and that worked too. In my test Boot State is OFF for all pins.

Maybe you can give me more information about your test setup. Or perhaps there was a bug in my "in progress" code that I subsequently fixed.

Mike

peoyli, 2023-02-14:
My test setup is actually nothing more than the HW-584 directly connected to a (out of the box compatible) 16 channel relay board with the usual '/822' pinout.
For the settings, I have all the pins inverted and 'boot state' set to OFF.

nielsonm236, 2023-02-14:
The pin mapping tables look fine. And the only place in the code where the pins are actually written is in main.c function "void write_output_pins(void)". At all other places in the code the pin states are managed in variables pin_control[]. So the physical pins are never written anywhere else EXCEPT for:
Pin 11, if used for the UART
Pins 14 and 15, if used for I2C
Pin 16, if used for DS18B20

Yes, there are several places in the code where the pin_control bytes are checked and may be over-written if a configuration conflict arises. For instance, if the user has DS18B20 checks and Pin 16 is defined as an output, the output definition will be overwritten. But this kind of cheeck and over-ride does not directly write the physical pin.

So if all we've done is change from /821 to /822, all Feature boxes are NOT checked, and the only place the mapping tables are interpreted and the physical pins are written is in function write_output_pins(), then it should work fine. It's working correctly in my test set up so there is a bug in the "interim" code you cloned. But that bug should show up as a problem in BOTH the /821 and /822 configurations ... the mapping tables are run through the exact same interpreter (write_output_pins()).

Mike

peoyli, 2023-02-15:
So if it's only in the interim code I compiled, it should be gone by release time.. but even more confusing, it's gone now (when I went from /821 back to /822 again). I have not reflashed the controller, nor changed any of the settings. I'm also pretty sure I have tested going between the pinout settings before (not only from 'default' value from writing the code to the controller, to /822).

peoyli, 2023-02-15:
I guess I'm onto something... I power-cycled the controller, and when it came up it didn't respond to any of the commands (tested both direct with the REST commands and from the web browser, web page was responsive as before but did not trigger any relay).. Did a /821 (change from /822) and it started responding again.

Another power-cycle (still in /821), not responding to anything when up again (iocontrol status shows that the outputs are on, but they do not change the relays). /821 woke it up, and /822 also works as expected (when running it after a cold boot)..

nielsonm236, 2023-02-16:
There could be any number of issues present. There was a lot of untested code mixed in with that interim code save. I only uploaded to Github to prevent loss of the code should my laptop crash. I was pretty sure the alternate pinout code was stable, but I was in the process of implementing the rather extensive changes to handle the PCF8574 and it is possible some instability was introduced.

peoyli, 2023-02-17:
I get a similar problem with a self-built browser only from the current code in the master branch.
The IOControl page indicates outputs are on, but turning on/off does not affect the relays.
This is after a cold start.When I reboot the network module using /91 (or the web ui) it starts working.

nielsonm236, 2023-02-17:
OK - thank you for the observation. Once I get the current changes stabilized we can jointly re-test for these issues.
Mike

nielsonm236, 2023-03-04:
The new code is stable now and I've tested the above. I'm unable to reproduce so you'll have to re-test once I get the Pre-Release out.

nielsonm236, 2023-03-05:
While walking through the various builds for retest I think I've reproduced this. It happens ONLY after programming new code into the module, where physical pin 16 gets stuck "ON". After a reboot the problem never occurs again (until I load new code again). For me physical pin 16 is the one that gets stuck "ON" regardless of the Pinout Option I choose. This is puzzling because after new code is loaded the firmware goes through the same "reboot" processes as occur when the reboot button is pressed in the GUI. So, this will take some debug time as I can't operate the UART during the reboot initialization.

nielsonm236, 2023-03-05:
With some more testing I've found this can actually affect other pins depending on which code load and test scenario I performed prior to loading new code. But I think I found the problem in the order of code execution that initializes the io_map_offset variable that points into the io_map table when multiple pinout options are supported. I've applied a fix but won't be able to test until tomorrow.

nielsonm236, 2023-03-05:
Tested and looks like it is fixed. This will be part of the pre-release.

nielsonm236, 2023-03-07:
Addressed in 20230305 1816

closed 2023-03-07 by nielsonm236