nielsonm236/NetMod-ServerApp #146: Incorrect content-length for most pages returning content

peoyli, 2023-02-27:
With the 'Browser Only' version from the current release (20230125), the Content-Length header returned using commands /60 - /63 and /71 mismatch with the actual number of bytes of content returned.

For example, curl (curl http://192.168.1.4:8080/63 >/dev/null) reports:
curl: (18) transfer closed with 1 bytes remaining to read

nielsonm236, 2023-03-04:
I didn't attempt to reproduce this in the 20230125 release, however in the current "in development" version I've verified that the problem does not occur. New Pre-Release should be out within a few days if I don't find any more bugs.

nielsonm236, 2023-03-07:
Addressed in 20230305 1816

peoyli, 2023-03-07:
Still the same with self-built (only changed build type to 'browser only' before compiling) of the current code:
```
root@itlab:~# curl http://192.168.1.4:8080/71 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  27876      0 --:--:-- --:--:-- --:--:-- 27876
curl: (18) transfer closed with 1 bytes remaining to read

```

nielsonm236, 2023-03-08:
I'm not sure what you are doing.
/71 only works in special debug builds. As such it should return the IOControl page (because it isn't enabled in normal builds). Here is what I get:
[image]

nielsonm236, 2023-03-08:
I'm trying to think of what might be different in your build environment, particularly since you received a dramatically different size for the response to the /71 command. You could try loading the NetworkModule-Browser.sx file and see what result you get. If it works that would indicate the self-build has an issue (although I can't imagine what ... yet).

nielsonm236, 2023-03-08:
While trying to research what might be going on I loaded the NetworkModule-Browser.sx file from the 20230125 2003 release. I do not see the one byte difference there either. So I think whatever problem that is occurrng in the self-built code has been there since that time. Are you building from the code in the .zip file with the release, or do you have some of your own modifications?
Also if you could send me a screen shot of your Configuration page that would help me to exactly duplicate your test. Configuration settings will change the content of the IOControl page, and it is possible that one of the settings is not properly accounted for in the page sizing. You can send the screen shot by email if you are concerned about showing it here.
Mike

peoyli, 2023-03-08:
> I'm not sure what you are doing.
> /71 only works in special debug builds. As such it should return the IOControl page (because it isn't enabled in normal builds). Here is what I get:

I used /71 (and happened to have that output on the last test) just for testing/verifying that the issue still exist. It's reasonable that /71 returned the control page. Here's the full output of my tests:

```
root@itlab:~# curl http://192.168.1.4:8080/63 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  27876      0 --:--:-- --:--:-- --:--:-- 27876
curl: (18) transfer closed with 1 bytes remaining to read
root@itlab:~# curl http://192.168.1.4:8080/60 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  27876      0 --:--:-- --:--:-- --:--:-- 27876
curl: (18) transfer closed with 1 bytes remaining to read
root@itlab:~# curl http://192.168.1.4:8080/61 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  4979   99  4978    0     0  31506      0 --:--:-- --:--:-- --:--:-- 31506
curl: (18) transfer closed with 1 bytes remaining to read
root@itlab:~# curl http://192.168.1.4:8080/62 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  28193      0 --:--:-- --:--:-- --:--:-- 27876
curl: (18) transfer closed with 1 bytes remaining to read
root@itlab:~# curl http://192.168.1.4:8080/63 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  27876      0 --:--:-- --:--:-- --:--:-- 27876
curl: (18) transfer closed with 1 bytes remaining to read
root@itlab:~# curl http://192.168.1.4:8080/71 >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  27876      0 --:--:-- --:--:-- --:--:-- 27876
curl: (18) transfer closed with 1 bytes remaining to read

```

nielsonm236, 2023-03-08:
Re-opening as discussion continues.
I am not able to reproduce this at all, so I need to know what the Configuration page settings are when it happens, in particular what Features are enabled, Device name, and IO Names. Those can change the size of the transmitted IOControl page and perhaps there is a bug in that code.

nielsonm236, 2023-03-08:
Ooops. Needed to click Reopen.

nielsonm236, 2023-03-08:
Can you run the Browser .sx file delivered in the pre-release package and let me know if you get the same results?

peoyli, 2023-03-08:
Just tested. Returns that same notice about the missing byte.

nielsonm236, 2023-03-08:
OK - Can you email the .sx file you built to me?  email to nielsonm.projects@gmail.com

nielsonm236, 2023-03-09:
Received the file. On the same module I ran my Browser .sx file and did not have a byte count mismatch. Then I loaded your file, got the same page transmission size, and no byte mismatch. So, I would say my .sx file and yours are identical at least with respect to the IOControl page transmission.

So, now let's try to figure out why your test process shows an issue and mine does not.

I'm running Windows 10 Powershell with a Curl command "Invoke-WebRequest -v http://192.168.1.184/71"
Are you running Windows? Or Linux?
One difference is that I run the Curl command manually one time, whereas your test runs it several times in series as part of a script. That shouldn't matter, but it is a difference. If you manually run your Curl command once from a command line do you get the error?
Are you able to run Windows Powershell to see if Curl produces the error in that environment?

I really don't think there is a problem in the Network Module code, but clearly we get different results and now it appears to be something in the way tests are run.

nielsonm236, 2023-03-11:
To continue pursuing this I created an Ubuntu Linux installation in Windows WSL. I ran the curl commands you used to see if perhaps the difference was between running curl commands in Windows Powershell vs running them in Linux. I did not have a byte count mismatch.
A suggestion: If you run the command without the >/dev/null the command should show the raw page content. That might show if a character is being dropped or added at the end of the HTML transfer.
Othewise I'm completely stumped as to why you are seeing this issue.
Mike

peoyli, 2023-03-11:
As a continuation to the reply I resent by email (as Github censored it), I have now found a way to reproduce the issue.

**Before:**
```
root@itlab:/home# curl -O http://192.168.1.4:8080/60
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  25843      0 --:--:-- --:--:-- --:--:-- 26115
curl: (18) transfer closed with 1 bytes remaining to read
root@itlab:/home# ls -l 60
-rw-r--r-- 1 root root 2481 Mar 11 12:33 60
```
**After long-press (10 sec+) on the reset button to go back to defaults (then before this test, set all to 'output')**
```
root@itlab:/home# curl -O http://192.168.1.4:8080/60
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2497  100  2497    0     0   6916      0 --:--:-- --:--:-- --:--:--  6916
```
**After configuration change: set all to 'invert'**
```
root@itlab:/home# curl -O http://192.168.1.4:8080/60
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2497  100  2497    0     0   6289      0 --:--:-- --:--:-- --:--:--  6289
```
**After configuration change: set IO names to K01 to K16**
```
root@itlab:/home# curl -O http://192.168.1.4:8080/60
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 99  2482   99  2481    0     0  25843      0 --:--:-- --:--:-- --:--:-- 25843
curl: (18) transfer closed with 1 bytes remaining to read
```
The problem lies within the calculation of the Content-Length (not with the content output itself), since 2497 bytes is correct for IOs named "IO01"-"IO16", and the change I do will cut each of the names by one character, so 2497-16 = 2481.

nielsonm236, 2023-03-11:
With your additional investigation I found the bug, and you are exactly right. The calculation for page size for Browser Only pages is incorrect. It leaves out the size of the last IO Name. This gets hidden a bit as most Browsers don't care about the final <body> and <HTML> tags, so the display will still be fine for many names even if the page size is incorrect. You can see this effect by varying the size of the IO Name for pin 16 and viewing the page size and the page content via curl. 

I've put a fix in. and will send a test file by email. I have one remaining concern: The rate at which IO change and Status Request messages are sent MIGHT be a problem - but in the information you've sent so far I'm not seeing an issue so we are probably okay.

nielsonm236, 2023-03-11:
@peoyli Files sent to you by email. If you think it is working now I will update the pre-release.
By the way, this bug has been there a very long time, perhaps since I first introduced the ability to apply names to pins (probably April 2021).

peoyli, 2023-03-11:
Just tested the browser-only build you sent, and I do no longer get the stderr output about the missing byte (compared to content-length).
I guess it's safe to say that this bug has been eliminated now when you found the problem and corrected it.

nielsonm236, 2023-03-12:
Addressed in General Release 20230312 1441

closed 2023-03-12 by nielsonm236