nielsonm236/NetMod-ServerApp #147: Question for team: How to handle the MQTT state-req message with PCF8574

nielsonm236, 2023-03-04:
My brain is fried so I'll throw this one out there for comment;

First, what I did with REST commands is that /98 or /99 report 16 pin state digits if no PCF8574 is found. But /98 or /99 report 24 pin state digits if a PCF8574 IS found. I leave it up to the user to detect this is whatever REST interpreter they are using.

Now the question: What do I do with the MQTT state-req command. It current returns two text bytes with the pin states of the 16 pins on the STM8 processor. But what if a PCF8574 is added. Should I return 3 bytes? Or should a new state-req command be added that returns 3 bytes? I'm asking because I'm not sure what complication this creates for MQTT software ... probably not any more complicated than the REST logic. I've never written an MQTT Client app so I'm looking for some suggestions.

For the moment the Pre-Release will never report the PCF8574 pin states when a state-req PUBLISH is received. It will operate as it does now reporting only the 16 pins on the STM8.

Mike

yozik04, 2023-03-04:
It will definitely break current clients if you will send 3 bytes instead of 2. =)

nielsonm236, 2023-03-05:
I agree. Just wasn't sure due to lack of experience with clients. I'll implement a new command. Perhaps state-req2? Unless someone has a better idea.
Just FYI, all the recent code changes (and bug fixes) have pushed the non-upgradeable MQTT version to essentially zero space. I will have to take a break from anything new to focus on code reductions for awhile.

nielsonm236, 2023-03-24:
Addressed in release 20230324 1435 by adding MQTT command "state-req24" and response "state24".

closed 2023-03-24 by nielsonm236