nielsonm236/NetMod-ServerApp #152: Short status should respect the setting of the "invert" option

peoyli, 2023-03-08:
Just a note, needs more testing.

Discovered this yesterday (running a test-loop with the newly implemented /51, but it's the same with the 'normal' short status /99)

When changing IOs to 'disabled' the status for that output is returned in a non-inverted state even if that IO was set to 'invert' before disabling it (OR the state the IO was last set to, even if disabled)

I will write a proper test in the upcoming days, but here's some output and explanation of what I'm doing in the test program:
Line 1 of output:
I have a set of files (text) which contains the reverse binary order for each of the relays I'm controlling, so the content of these setting-files could be a full 16-bit binary number with zeroes for 'off' and ones for 'on'. These could be full length or short (as in '1011' to only set the status of outputs 1-4, bits 0-3).
Line 2: I reverse the content of that file
Line 3: then pad it with zeroes to make it full-length
(quietly converted to hex and sent to the controller)
Line 4 (or 5 when I added a line break to make it more readable): status returned by /51 (same as /99)
After a completed run, I send /56 to turn off all IOs

Test A, all relays enabled, 'invert' mode (active low board connected)
```
root@itlab:/home# php test.php
Relay group: 0
1110000000000000
0000000000000111
0000000000000111
status: 0000000000000111
Relay group: 1
0100000000000000
0000000000000010
0000000000000010
status: 0000000000000010
Relay group: 2
0010000000000000
0000000000000100
0000000000000100
status: 0000000000000100
Relay group: 3
0001000000000000
0000000000001000
0000000000001000
status: 0000000000001000
Relay group: 4
0000100000000000
0000000000010000
0000000000010000
status: 0000000000010000
Relay group: 5
0000010000000000
0000000000100000
0000000000100000
status: 0000000000100000
Relay group: 6
0000001000000000
0000000001000000
0000000001000000
status: 0000000001000000
Relay group: 7
0000000100000000
0000000010000000
0000000010000000
status: 0000000010000000
Relay group: 8
0000000010000000
0000000100000000
0000000100000000
status: 0000000100000000
Relay group: 9
1110000000000000
0000000000000111
0000000000000111
status: 0000000000000111
Relay group: 10
0001110000000000
0000000000111000
0000000000111000
status: 0000000000111000
Relay group: 11
0000001110000000
0000000111000000
0000000111000000
status: 0000000111000000
Relay group: 12
111
111
0000000000000111
status: 0000000000000111
Relay group: 13
110
011
0000000000000011
status: 0000000000000011
Relay group: 14
0111
1110
0000000000001110
status: 0000000000001110

```

Test B, IO 16 disabled, other enabled and set to 'invert', stopped to also disable IO 01 (Test C)
Status shows IO 16 as 'on', no matter what I set/change it to during the test.

```
root@itlab:/home# php test.php
Relay group: 0
1110000000000000
0000000000000111
0000000000000111
status: 1000000000000111
Relay group: 1
0100000000000000
0000000000000010
0000000000000010
status: 1000000000000010
Relay group: 2
0010000000000000
0000000000000100
0000000000000100
status: 1000000000000100
Relay group: 3
0001000000000000
0000000000001000
0000000000001000
status: 1000000000001000
Relay group: 4
0000100000000000
0000000000010000
0000000000010000
status: 1000000000010000
Relay group: 5
0000010000000000
0000000000100000
0000000000100000
status: 1000000000100000
Relay group: 6
0000001000000000
0000000001000000
0000000001000000
status: 1000000001000000
Relay group: 7
0000000100000000
0000000010000000
0000000010000000
status: 1000000010000000
Relay group: 8
0000000010000000
0000000100000000
0000000100000000
status: 1000000100000000
^C
```

Test C: IO 01 (and 16) disabled, cleared all outputs before running, status shows IO 01 as 'on', no matter what I set/change it to during the test. Stopped running it to add line break before the returned status (Test D).
```
root@itlab:/home# curl http://192.168.1.4:8080/56
root@itlab:/home# php test.php
Relay group: 0
1110000000000000
0000000000000111
0000000000000111
status: 0000000000000111
Relay group: 1
0100000000000000
0000000000000010
0000000000000010
status: 0000000000000011
Relay group: 2
0010000000000000
0000000000000100
0000000000000100
status: 0000000000000101
Relay group: 3
0001000000000000
0000000000001000
0000000000001000
status: 0000000000001001
Relay group: 4
0000100000000000
0000000000010000
0000000000010000
status: 0000000000010001
Relay group: 5
0000010000000000
0000000000100000
0000000000100000
status: 0000000000100001
^C
root@itlab:/home# curl http://192.168.1.4:8080/56

```

Test D: IO 01 (and 16) disabled, cleared all outputs before running, status shows IO 01 as 'on', no matter what I set/change it to during the test. Stopped running it to add line break before the returned status (Test D).

```
root@itlab:/home# nano -w test.php
root@itlab:/home# php test.php
Relay group: 0
1110000000000000
0000000000000111
0000000000000111
status:
0000000000000111
Relay group: 1
0100000000000000
0000000000000010
0000000000000010
status:
0000000000000011
Relay group: 2
0010000000000000
0000000000000100
0000000000000100
status:
0000000000000101
Relay group: 3
0001000000000000
0000000000001000
0000000000001000
status:
0000000000001001
Relay group: 4
0000100000000000
0000000000010000
0000000000010000
status:
0000000000010001
Relay group: 5
0000010000000000
0000000000100000
0000000000100000
status:
0000000000100001
Relay group: 6
0000001000000000
0000000001000000
0000000001000000
status:
0000000001000001
Relay group: 7
0000000100000000
0000000010000000
0000000010000000
status:
0000000010000001
Relay group: 8
0000000010000000
0000000100000000
0000000100000000
status:
0000000100000001
Relay group: 9
1110000000000000
0000000000000111
0000000000000111
status:
0000000000000111
Relay group: 10
0001110000000000
0000000000111000
0000000000111000
status:
0000000000111001
Relay group: 11
0000001110000000
0000000111000000
0000000111000000
status:
0000000111000001
Relay group: 12
111
111
0000000000000111
status:
0000000000000111
Relay group: 13
110
011
0000000000000011
status:
0000000000000011
Relay group: 14
0111
1110
0000000000001110
status:
0000000000001111
root@itlab:/home#

```

nielsonm236, 2023-03-08:
I could be misunderstanding the issue.
If a pin is disabled what it reports shouldn't matter regardless of any prior state, other setting, or change on the pin. The title of the Issue is a bit confusing as Status does respect the setting of the "invert" option, EXCEPT when a pin is Disabled. In that case the code has no history to determine with certainty what state should appear in the Short Form report. The code no longer knows if the pin was formerly an Input, Output, Linked, was a 0 or a 1, what the former boot state was, or if it was previously Inverted. Any of those settings could have been changed at the same time that the pin was set to Disabled, so sorting that out becomes impossible.
What do you want Status to report when a pin is Disabled? We would have to define that, considering all possible prior settings, to make any change to the code. And sorting that out might be difficult (meaning much code) depending on the requirement.

If you find that Invert is not being applied when a pin is set to Input, Output, or Linked I need to know about that. But also keep in mind that at the point a pin Type is changed there is no history. Only what is left behind in the pin_control variable (if an Output) or what is currently on the pin (if an Input).

nielsonm236, 2023-03-12:
Addressed in General Release 20230312 1441

closed 2023-03-12 by nielsonm236