nielsonm236/NetMod-ServerApp #162: Intermittend reboot netmodule

sjorsBe, 2023-03-25:
Dear all, 
Im working with the linked option right now but in the last releases i experience a spontaneous reboot of the module a few times a day. The effect is that the linked input and output loose their connection with home assistant and during reboot change their settings triggering home assistant to switch on ( or off) a number of devices logically connected to the modules pins. 
It happens that in the middle of the night a numer of lights switch on due to such a spontaneous reboot ( quite aanoying :) ) 
Do others also experience these spontaneous reboots and if so, is this due a failure in the software or someware else?

Kind regards 

Sjors

nielsonm236, 2023-03-25:
A few questions:
- The Linked Pin feature first came out in the 20221217 release. Has the problem you are seeing been there since the feature first came out? Or only in the last couple of releases?
- Do you have more than one module running and are all modules having the same problem?

Since you are seeing it multiple times a day I will add some debug to my lab setups to see if I observe the same behavior. Typically spontaneous reboot is power related, but there is also a hardware opcode/timeout watchdog that might be firing. We've added a huge amount of code in the last three months so there might be a problem there. But when a problem is spread out by several hours I get very suspicious of power glitches.

Would be good to hear if anyone else is seeing a problem while I set up the lab monitoring.

Mike

nielsonm236, 2023-03-25:
I've set up my lab configurations as shown here. All outputs are ON, with their "boot state" set to OFF. If any of them reboot the outputs will turn off. I will monitor this today to see what happens.
[image]

nielsonm236, 2023-03-25:
While this continues to run (2 hours so far with no issue) I have a few more questions that can help me set up a better test if the current test doesn't show anything:
- In the location(s) showing a problem, is the PCF8574 being used?
- From your description it sounds like you are using MQTT Upgradeable FW. Is this correct?
- Do you have a polling process running in Home Assistant? Or just letting the Network Modules report changes?
- Do you have any other polling processes running, perhaps polling with REST commands? Or running NodeRed for polling?
- How many pairs of Linked pins do you have?
- Are you using Power Over Ethernet? If yes what do you use as a power source and end-point power converter?

Most of the answers to the above shouldn't make any difference, but might result in different code paths being exercised, or different run timing. So I'm just looking for ways to more closely emulate your setup and hopefully provoke the problem.

Mike

nielsonm236, 2023-03-26:
No resets in the first 13 hours on the nine units. I will check again in the morning.

nielsonm236, 2023-03-26:
I'm at 24 hours. No resets on any of the 9 boards. @sjorsBe if you can provide any more information (see questions above) I will add it to the test. But at this point I think you should look into power line glitches getting through to the HW-584 input power. The reset circuit on the HW-584 is very sensitive.
Mike

jmcvieira1, 2023-03-26:
Yesterday I also put a module in test with pcf and ds18b20 so far no reboot
J. Vieira

nielsonm236, 2023-03-26:
@sjorsBe Another possible solution for you: Since your application is for room lighting I imagine the lights are only turned on and off a few times per day. You could experiment with setting the outputs to "Retain" so that if a reboot occurs they should retain their previous state. Having said that, I have two comments;
a) You really don't want the intermittent resets to occur at all, so I still think looking at power is important.
b) I haven't tested power loss with Linked Pins and Retain, so there might be an issue that needs to be addressed. For instance, say there is a power loss, and the input switch is changed while power is down. What happens to the output when power returns? Because I have not previously tested the combinations I'm a little concerned about how the boot code handles the state of the Linked Input even though the Linked Output is set to Retain. I won't have time to test all the scenarios right away so you can decide what you want to look at.
Mike

sjorsBe, 2023-03-26:
Hi Mike 
I was not able to email yesterday so I try to make up for it today with a more clear description and an answer to all your questions:

**Background info:** 

The last number of weeks I moved from my combined Homebridge/ Homekit environment to the Home Assistant environment for the home automation solution.  Up until then the netmodule in use was in experimental mode to control a number of test items. 
The  netmodule was moved to “production” which  is the home assistant environment.
The hardware structure of the module is: 
-	The module received its own power supply ( 5V 1A ) + elco buffer. 
-	On pin 16 are 4 DS18B20 temp sensors, Pin 8 is disabled 
-	Pin 1 is in linked mode to pin 9
-	Pin2 is in linked mode to Pin 10
-	Pin 3 is in linked mode to Pin 11
-	Pin 4 is in linked mode to Pin 12
-	Pin 5-7 are in input mode 
-	Pin 13-15 are in output mode 
Module was set to HA auto, MQTT and DS18B20 “on”

The firmware used is the MQTT firmware, Updated with the ST link, No extra EEPROM, No PCF8574
The firmware number is 20230312 MQTT ----

Pin 9-12 are connected to 4 solid state relays,( one per pin)  consuming 2mA each ( and working fine Pin 5 is connected to a wall-switch, switching to 5v input or 0V input as logical 1 and 0. Corresponding pins 1-4 can switch from 0v to 5V.

System runs fine, MQTT works ok, switch starts automations in home assistant turning  on an off the attached Zigbee lights

Running this configuration Home assistant log from pin 5 shows on average 2 times a day: 
“Input 05 is not available anymore” If I check other pins the message and timestamp is the same. 

This means that the netmodule is rebooting and falls back to the bootup settings. This switches the automations within Home Assistant to either switch on of switch off the attached Zigbee lights. 

The question is: Why does this rebooting happen?

Could it be: 
-	Overheating of the module ?
        I recorded the heat of the heatsink of the netmodule: on average 24-26 degrees Celsius. 
-	Instable power supply:
        I buffered the power supply with a 1000uF elco. 
        The rebooting still happens. 

Possible extra tests:
-	Switching PSU. 
        That experiment is now running 
-	A problem in the hardware or software in the module itself. 
        That is what you guys are checking now.
-	If nothing delivers a solution I will switch the entire module for another one. 

Im open for all different ideas!!

**Answer to questions:**  

1.	The Linked Pin feature first came out in the 20221217 release. Has the problem you are seeing been there since the feature first came out? Or only in the last couple of **releases?**
Answer: I do not know. Since the earlier use of the modules was short time experimental, there was no opportunity to witness this problem.  Also: the combination HOmebridge/Homekit did not have the option to record logs. 

**2.	Do you have more than one module running and are all modules having the same problem?**
Answer: Im now working with one module, since it is integrated together with the solid state relays and PSU in one specially made casing. If the problem persists, I have to disassemble the entire case and rebuild it with a new module. 

**3.	In the location(s) showing a problem, is the PCF8574 being used?**
Answer:  No it is not, not even installed. 

**4.	From your description it sounds like you are using MQTT Upgradeable FW. Is this correct?**
Answer:  No this is not correct. For this solution I started with the simplest module config: An MQTT firmware, updated with the ST-Link tool. 

**5.	Do you have a polling process running in Home Assistant? Or just letting the Network Modules report changes?**
Answer: I don’t understand the question but Im using the logfiles reporting within Home Assistant all activity of the Netmodules Input05.

**6.	Do you have any other polling processes running, perhaps polling with REST commands? Or running NodeRed for polling?**
Answer:  No. 

**7.	How many pairs of Linked pins do you have?**
Answer: 4 linked pairs (see background info )

**8.	Are you using Power Over Ethernet? If yes what do you use as a power source and end-point power converter?**
Answer:  No, due to the fact that the connected switch must be functional when the connection between netmodule and LAN swich is severed. When I would have used POE the module would not have power whenever the connection with the switch was severed. 

I hope this is the extra info you need and thanks to all people who are helping to fix this!!

Kind regards 

Sjors

closed 2023-03-26 by sjorsBe

jmcvieira1, 2023-03-26:
In all the modules that I have in operation when I want to use an IO as input I always use a HEF4050BP or similar, this ic allows converting an input signal from 3.3 to 15V to an output signal of 3.3V, I know that the networkmodule is 5V tolerant but I don't know what its reaction will be being exposed for hours to one or more 5V inputs maybe Mike has some experience in this matter.

J. Vieira

nielsonm236, 2023-03-26:
@sjorsBe With regard to power filtering the problem may be a spike or ringing as the result of a spike. A couple of 0.1 uF ceramic caps in parallel with the 1000uf cap may help.
Even though the 5V 1A power supply should be more than enough current, you may want to check the output voltage when it is attached to make sure it is not "drooping".
In answer to @jmcvieira1 Sjors should be OK as he is driving solid state relays. I suppose reviewing the specs to make sure they are being adequately driven would be a good idea, in particular making sure they can't inadvertently cause a large current draw on the drive signals which could cause a power glitch.
@sjorsBe Is there any indication in the logs that the reboot occurs around a point in time that the outputs are switched? I'm trying to think of the various things that I've seen cause power issues: a: Lower than expected input voltage, b) High frequency spikes on the A/C line making it to the 5V output, c) Load switching that causes a very short (high freqency) spike on the power rails.
Mike

nielsonm236, 2023-03-26:
@sjorsBe When you get a chance can you post the part number for the solid state relay you are using? Thanks, Mike

sjorsBe, 2023-03-27:
@nielsonm236 Its a bit difficult, the solid states are at the bottem of the casing. 
But it is a standard module also used to switch 230v AC with an arduino - from ali express. 
The board has 4 solid states on it with active led signaling and protective resistors in place. 
It has its own power ( 5v) and a common ground with the module. Each Solid state relay has its own input trigger 

hope it helps

jmcvieira1, 2023-03-27:
Something like this...
[image]

sjorsBe, 2023-03-27:
@jmcvieira1 
more or less. but there is a range of quality differences on AE. I am using the low current ones

sjorsBe, 2023-03-27:
no, just some lights 230v AC - no leds, no drivers, no motors 
the input side are connected to wall switches. with a small resistor attached to minimise current. ( 680 ohm) 

the 5v side hardly uses current. So it can not overheat the Netmodule.

sjorsBe, 2023-03-27:
WHat i described was the current status of the box. 
Power comes in from top right side ethernet from top left side.
The output power (4x)  comes from right bottom  and the 5v switches wires come from left bottem side 
The box is 15 by 10 cm 

should be space enough. 

i now switched off the small 220v AC - 5v dc module and installed an external apple ipad power supply. See how it goes :)

jmcvieira1, 2023-03-27:
I didn't make myself understood sorry, what I was asking was the connection between the networkmodule and the wall switches you have in your house:
Where did you pass the wires to the wall switch, are they close to the ac voltage wires?
Do you have any wall switch with ac voltage next to the one you use with 5vdc to control the networkmodule?

sjorsBe, 2023-03-27:
I tried to explain.  
No they are not close to eachother.   The 220v is  located on one side of the casing and the 5v connections to the wallswitches on the opposite side

jmcvieira1, 2023-03-27:
so where are the wall switches physically located that you use to control the networkmodule they are all next to the casing with the networkmodule?

sjorsBe, 2023-03-27:
Grin, 

No,they are connected through dedicated cat6 wiring of max 4 meter through wallpipes. The old 230v wiring has been removed

jmcvieira1, 2023-03-27:
cat6 utp or stp?

sjorsBe, 2023-03-27:
Utp

jmcvieira1, 2023-03-27:
Could it be that in the vicinity of that cat6 cable there is another tube with ac voltage, this will disturb the signal and may cause the networkmodule to reboot, if you continue to have problems with the sporadic reboot of the networkmodule you can try unplugging all connections to the wall switches on the box next to the networkmodule and test.

jmcvieira1, 2023-03-27:
If there is one nearby i would advise you to use any cable as long as it is shielded, and connect the shield to negative and the negative to the ground off you house (green yellow wire).
But first it would be good to define exactly what is causing the networkmodule to reboot.