nielsonm236/NetMod-ServerApp #169: Request on reopening Intermittend reboot Netmodule

sjorsBe, 2023-04-05:
Hi All, 

as an addition to the now closed issue of Intermittend reboot Netmodule the following: 

The module kept on rebooting with an interval between 8 and 16 hours 
There was no link with activity of the module itself. 
Tests were done with 
- all outputs ON, half of outputs  ON, no outputs ON.
 - all inputs linked to 5v, inputs linked to 3 volts, inputs linked to ground 
 - 
The result was no change in the irradic behaviour of the netmodule. 
It kept on rebooting between 8-16 hrs 

So I changed the entire module but it had the same result with the other ones!
So I changed the power supply: 
- from simple DC PSU to well stabilized Lab PSU of 5v DC ( straight DC line on the scope. ) 
The problem was still there. 

So I hooked up a temp sensor to the small heatsink and there seems to be a relation whenever the heatsink temp reaches 30 degrees C.  This seems to cause the modules to reboot. 

So I now ordered a pure copper heatsink ( 10 x 16 * 11 mm)  and will glue it with proper heatconducting paste to the chip. 
See if that makes a difference. 

I wonder what is your experience with a proper build in module in a not cold environment?

Please let me know. 

I 'll keep you posted on the experiments. 

kr 

Sjors

nielsonm236, 2023-04-05:
I have 9 modules running constantly just for test.  Room temperature runs from about 21C (typically at night) to 28C (daytime).The room has other equipment running in it so it stays warmer than the house at night (our night thermostat is set to 17C.

Which chip is getting hot?  The STM8S processors on my are just a little warm. The ENC28J60 chips get "hot", but even without a heatsink I can keep my finger on them for a minute or more. Covering a bare ENC28J60 chip with my finger causes them to warm fairly quickly due to no air exposure. While the manufacturer didn't provide instructions I'm fairly sure the heatsink they provide with the boards is supposed to be on the ENC28J60.

I will see if I can warm up the test matrix somehow. Here's a photo of how I have the test boards set up: Maybe just hang a cover over them for self heating and monitor the temperature sensors on them.

[image]

As FYI there was one other person emailing me about resets a couple of month ago. I will have to go back and find those emails. He said his device ran fine for over a year, then started having problems. At the time I questioned temperature, but he said the location was room temperature. No actual temperature measured, so "room temperature" is subjective.

Mike

nielsonm236, 2023-04-05:
I have one DS18B20 on each of 5 modules. It is easy for me to add up to 5 on one of the modules. I doubt this is the problem, but it is good to try any possibility we think of. What pullup resistance do you have on the DS18B20 data line?

Another thing to consider: I know your relay specification is 2mA input. But do you have a pull up on those pins too? What value? When the pin is high, what voltage is on the pin? What I'm thinking about is if there is too much current being drawn when pulling the pin low.  OR if there is not enough pull-up on the pin then the relay might pull the pin to some voltage that causes the pin to draw current. This might heat up the STM8S.

Mike

nielsonm236, 2023-04-05:
I was just re-reading info provided in issue 116. I don't think the pins attached to the solid state relays are the problem. At least by the info you provided it should not be a problem.
Are you able to measure the total current drawn through the 5V to the HW-548? That might tell us if there is excess current draw somewhere on the board ... and if too much current we know to look for the source of that problem. I can measure a couple of my test boards to see what I have here.

jmcvieira1, 2023-04-05:
Three hours ago when I received notification of this problem I looked at my test bench and wrapped a rag around the networkmodule, after about three hours it still works without reboot.
the STM8S temp is  38,7ºC   and the lan controller is 51,9ºC
I have running one DS18B20 and one PCF
[image]
[image]
[image]
[image]

jmcvieira1, 2023-04-05:
Sjors to be really sure that the temperature is the origin of the problem, you can use a hair dryer and heat the board to see if it reboots.

nielsonm236, 2023-04-06:
I only got my boards up to about 27C last night, I need to figure out another way to warm them up.  Will work on that later today.

My mind is still stuck on power as the source of the problem since I've seen that so often. I know your are sourcing 5V from a lab supply, which is a good choice for this testing.

How is the lab power supply connected to the HW-584? Directly via the 5V screw terminals? Or through some other board? 
If the lab supply is connected to the 5V terminals how long is the wire and the approximate gauge of the wire?
If the power connects to a different board first (perhaps the relays) then goes to the HW-584 what is the gauge of the wire in that connection?

I made some mistakes with wire gauge and wire length in previous test configurations, which is why I'm asking. 

Mike

sjorsBe, 2023-04-06:
Hi Mike

I see a lot of questions, thanks for that
Im in the middle of a hectic period of work right now, I’ll try to answer you on Sunday is that ok?

Sjors

nielsonm236, 2023-04-06:
Yes that is fine. In the meantime I will continue to pursue the heat testing.

nielsonm236, 2023-04-06:
I started a test on 9 modules where the temperature is being maintained between 35.5 and 36.5 C.  The test started about 3 hours ago.
- One module has 5 DS18B20 sensors and a PCF8574.
- Four modules have a single DS18B20 sensor.
- One module has the BME280 sensor.
- Three modules have no sensor.
- None of the devices on my boards have heat sinks.
So far no reboots but Sjors experience suggests this should take up to 16 hours. I'll see how they are running over the next 24 hours.

nielsonm236, 2023-04-07:
The test has run about 16 hours now. No resets. This is the recorded temperature from the DS18B20 recording the lowest temperature. The others look about the same but up to 2F higher.
[image]
No resets and no significant complaints recorded in the Link Error Statistics (a couple of MQTT traffic retires but that is normal).
As I mentioned before I have no heatsinks or air circulation so the device temperatures are far above what is being measured by the DS18B20's. I'm taking a risk of damaging the devices but will end this test in about 6 hours.
Mike

nielsonm236, 2023-04-07:
I'm not sure they are being damaged, butit is a risk. :-)
This test was worth it to see if there was a temperature related problem.
I'm not home right now but I logged in to see if they are still running.  All still OK.
Mike

nielsonm236, 2023-04-07:
Ended the test at 21hr 30min. Average temperature across all DS18B20's is 99.8F / 37.6C. I will leave them running as they cools to see if that causes a problem.
And correcting a mis-statement: 2 out of 9 of the ENC28J60 devices did have heatsinks. The rest did not.
Mike

nielsonm236, 2023-04-08:
No problems during the cool down.
[image]

nielsonm236, 2023-04-15:
@sjorsBe Any progress? Have you tried the firmware that I emailed that gives more detail about boot source in the Link Error Status?
Thanks
Mike

jmcvieira1, 2023-04-15:
@nielsonm236  
Hi Mike, I've been busy and it's only now at the end of the week that I put the test firmware to test after 48h these are the results:
Code Revision 20230405 2055 Browser UPG 
Link Error Statistics
31 0000135998
32 0001247103
33 0000060000
34 0000000000
35 0000000000

Code Revision 20230405 2055 MQTT UPG ...
Link Error Statistics
31 0000135710
32 0000547269
33 0000060000
34 0000000000
35 0000000000

nielsonm236, 2023-04-15:
@jmcvieira1 Thanks. Line 34 indicates no reboots (expected). Will be interesting to see what Sjors sees if he can reproduce the reboot problem. The biggest problem will be that a reboot still occurs but without a count in Line 34. That would strongly indicate a power loss. It is possible for a counter to increment with a power loss if the loss is very slow (like a very slow voltage drop off with noise on the power which might cause restarts without completing boot ... and thus improper processor execution). But that is unlikely. Usually a power loss is quick and no illegal operation count occurs, mostly because the power loss detection in hardware has fairly good hysteresis.

nielsonm236, 2024-06-01:
Closed due to lack of activity.

closed 2024-06-01 by nielsonm236