nielsonm236/NetMod-ServerApp #175: Save and Refresh have stopped working on Chrome?

nielsonm236, 2023-05-23:
This morning I noticed that Chrome freezes if I click on Save or Refresh on the IOControl page. I see that Chrome 113.0.5672.127 released on May 16 2023, and my system automatically updates Chrome.

The same Network Modules continue to respond just fine with Edge and Firefox, even while the Chrome window is still frozen. When I run Developer Tools with Chrome it looks like the POST is sent from Chrome, but when I check the IOControl page with Firefox I don't see that the POST was executed - like it was not received at the Network Module from Chrome.

Is anyone else seeing this issue?

nielsonm236, 2023-05-23:
Thanks. I think I know where to start looking.
Mike

kiralikbeyin, 2023-05-24:
While working with the ESP8266 chip, I encountered a situation where the request did not end and there was a timeout at the end because I did not close the connection after the client request in the web browser. I think there is an issue with that. Pay attention to the content length. Make sure the connection is completely closed.

If necessary, instead of satisfying the request inside a loop; A solution might be to run the client in a task and then delete the task completely at the end. (Maximum 5 clients could connect in the Esp8266 library.)

I haven't looked at your Github codes, just a few ideas..

kiralikbeyin, 2023-05-24:
Sometimes it's easier to rewrite the code than to find the bug and fix it. I think you can solve it quickly with GET method instead of POST method.

nielsonm236, 2023-05-24:
Yeah ... that's a lot of re-writing. 
In general I agree with you but have not undertaken the task due to the time it would take and the concern that it might end up taking more code space than the current POST processing. However: a) The POST processing code is quite large, so perhaps GET would be smaller, b) It would likely be less error prone, c) It requires re-write of both the C-code and the HTML/Javascript. No rocket science, just work.
As FYI I would still need POST code for the upgradeable versions to transfer new executable files to the server. But that is mostly a separate set of routines so the separation of data-POST and file-POST would not be too difficult.
I will put some more thought into this. I can already sense some users thinking "Ugh ... he's going to tear up the code again".

nielsonm236, 2023-05-25:
@kiralikbeyin Could you private message me at nielsonm.projects@gmail.com? I'd like to ask a couple of questions about using GET in this case. Perhaps I can draw on your experience to build some code for an experiment.

nielsonm236, 2023-05-28:
I haven't heard back from @kiralikbeyin, but I think I've got this resolved. I found a logic issue in the parsing process. Testing this is very difficult because the failures are always corner cases that alias with packet boundaries (specifically the location of the &'s in a POST relative to the packet boundaries). So, I'm going to do more testing before I release the fix.

@jmcvieira1 I wasn't able to reproduce your report, but I suspect it was the same bug. My inability to reproduce can easily be some change in packet boundaries. HOWEVER - while trying to reproduce I had the PCF8574 appear, then disappear, and I could only get it to reappear with a reset (I was using the reset button on the module to do this). I suspect it may have just been poor wiring in my attachment of the PCF8574, but I will make several more attempts to figure out what happened.

I am intrigued by the suggestion from @kiralikbeyin . I think Jevgeni suggested it once before. But, I am not sure how to implement the Javascript to perform multiple GET requests instead of one POST. One of the POSTs returns over 600 characters, and GET requests are limited to 255 characters, so it might take three or more GET requests to return the same data contained in a POST. Additionally we could end up with some new aliasing problems as follows; In a POST I'm able to wait until the POST completes before processing the configuration changes, which enables me to check for conflicts in the configuration. In the GET method I might have to come up with a new method of making sure all changes is a given session are reported before processing the changes. But my first impression is that the GET method might free up some Flash space.

nielsonm236, 2023-06-04:
Fixed in release 20230603 2058. There was one logic problem in POST parsing resulting from a change in parsing methods long ago that left a gap in detecting aliasing of the '&' delimiters when they occurred at packet boundaries. In addition there was an error in the POST size count for Configuration pages.

closed 2023-06-08 by nielsonm236