nielsonm236/NetMod-ServerApp #177: Need a command to initialize PCF8574 pins to a more benign state?

nielsonm236, 2023-06-03:
While performing test on Issue #175 and #176 I added a PCF8574 to a Network Module that had not previously had a PCF8574. The PCF8574 Configuration came up as
All pins are Outputs
All pins are Inverted
All pin Boot States are Retain
All pin Timers are 16383
All pin Timer Units are 1 Hour
While that could be a real Configuration, it came up that way probably because the I2C EEPROM locations where this information is stored were set to all 1's because they hadn't previously been used.

The only problem here is that this comes up looking rather odd to me as a user. The pins on the STM8S come up as Disabled, Not Inverted, Timers = 0, Boot State is OFF, Timer Units = 0.1s. I'm able to do that on a "fresh" Network Module because the STM8S doesn't have the "magic number" if never used, so I initialize everything. I'm not able to do that in an automated way with the PCF8574 because it is typically added later, and I have no way of knowing if it just appeared or it has been around for awhile.

I may add a URL command to "initialize" the PCF8574 pins to that more familiar look, although I need to be careful that this consumes only a few bytes of Flash. This is a "non-urgent" issue.

jmcvieira1, 2023-06-04:
I've already come across this behavior, and to avoid having to mess with all the settings I use command 74 of the code uploader to delete the eeprom content, and so I have a clean screen in all the setting's.

nielsonm236, 2023-06-05:
That is a good work around. Of course you have to remember re-upload the Strings and Firmware.
I think I'll keep the "initialize PCF8574" URL command on my to-do list as a more intuitive method for future implementation.