nielsonm236/NetMod-ServerApp #192: Logging

jimo42, 2023-08-04:
Hello. 
First, let me say that your firmware is great. I have bought the module some months ago and was very displeased by the functionality, especially the fact that it doesn't keep settings through restart. After buying the programmer, I have now what I was hoping for from the beginning, and it does what it should (which is, in connection with photovoltaic power plant on the roof, turning on 1 or 2 or 3 2-kW heating coils in water heater and curculation pump cycling, based on the current sun conditions and fullness of home battery.
So again, great help already!

OK, now for the idea/feature request.

What would I like:
To be able to have some logging.

Why: 
What happened me yesterday is that I've set on the relays at some point and let it be. Then, I have "reactive logging" - I am periodically checking status with /99 and writing it to a log file on home server, with timestamp. So I can see that after an hour and half, all relays went off, however I have no clue why. It could've been power outage, but I see none in the logs of FVE. I am suspecting my wife doing it accidentally as I had on her PC the page with /60 opened, but she said she just closed the tab, without even refreshing it (and even if so, refresh should not switch anything).
**So I was thinking if there would be an option for logging.**

How:
I guess there is no real "memory" in the module to keep logs (?). 
So maybe there could be an option to send logs to external server. Something basic, like POST it to given IP:PORT, or even GET it "to" there, encoded in URL.

Content:
Important info would be the command executed and source IP from where it came.
Timestamp can be added on the server side.

Thank you for considering.
--jimo42

nielsonm236, 2023-08-05:
@jimo42 You are correct that there isn't any real storage available for logging, unless you implemented the I2C EEPROM and ran an upgradeable version of the firmware. Then it might be possible, but might still be an issue with regard to size of the implementation.

With regard to posting to an external server: My firmware is a webserver, not a client. So I think posting to an external server would require adding Client firmware. I'm not sure how involved that would be, especially since some of the firmware versions have less than 50 bytes of Flash left.

What firmware type and version are you running?

As FYI, I've had a couple of other users complain about the hardware "resetting" or "rebooting" unexpectedly, with one person reporting the issue occurred daily. I have not been able to replicate that issue except in the case of power issues - and not even a visible power loss, but such things as an inadequate 5V supply on the module (where voltage drops to around 4.75V), or a glitch caused by sharing the same power supply with relays, and when a relay switched it caused the power to the HW-584 to glitch, resulting in a reset. The HW-584 design has a very sensitive power out detector, and even a momentary glitch can cause a reboot. But I've had 9 of the devices running in my test lab for months without seeing unexpected reboots. So I hate to say "I suspect a power issue" without a little more knowledge of your implementation.
Mike

nielsonm236, 2023-08-05:
Another thing you can do using your current logging method is to log the result of the /66 command ("Link Error Statistics"). It is more fully described in the Manual, but the bits to look for are:
1) Stack Overflow: If ever = 1 it would tell us that a code error occurred.
2) ILLOPF Counter: If ever > 0 there was an illegal opcode. This is likely a code error.
3) IWDGF Counter: If ever > 0 there was likely a code error.
While possible that a power fault could give one or more of the above, that is unlikely. So if you are seeing a reboot but none of the above occurred it increases the probability that a power glitch caused the reset.
You can reset the above indicators with the /67 command.

Reminder: Let me know the firmware type and revision (this is displayed at the bottom of the Configuration page),
Also: I would also like to know how often you are sending the /99 command (just a rough estimate).
Also: How often do you intentionally change the output states? (just a rough estimate).

Thanks
Mike

jimo42, 2023-08-22:
Hello Mike!
You are very reactive, thank you for that. I am sorry for being a bit "in the air" during the summer now, so will reply better in September. Quick msg, however - I think the sensitivity for input power might be the issue here (by the way, my module reseted once again since the first message). Yes, I have a 5V USB power adapter, used to power the module and also a module with 4 relays. Moreover I am using other outputs from the module to directly trigger another 3 high-power relays in the distributor box.
If you say the module is sensitive to even slight drop of the voltage, I think the first thing I will try is to add some big capacitor parallel to the source input and will see if that will help.
For other points, I will reply later. Honestly, if the capacitor will solve my problem, I won't have a need for logging anymore (or will just be ok with the "polling logging" I have already).

Thank you once again for the great work, your firmware rocks!

Jiri

nielsonm236, 2023-08-24:
The large capacitor will help with "power droop", but won't help if the problem is a high frequency spike. You may have to put some smaller ceramic caps in parallel with the large one. But try the large one first.
I always recommend that no relays be on the same 5V source as the Network Module, just because I've seen problems with that in my test setups. I now always put the Network Module one one source, and the Relays on another. I don't think this is a problem with solid state relays, but definitely a problem with coil type relays.
Mike

nielsonm236, 2024-06-10:
Closed due to inactivity.

closed 2024-06-10 by nielsonm236