nielsonm236/NetMod-ServerApp #200: HomeAssistant MQTT Breaking changes in 2023.8+

yozik04, 2023-08-22:
Hello @nielsonm236,

I trust you are well!

HomeAssistant has thrown some work at us.
See [Breaking changes](https://www.home-assistant.io/blog/2023/08/02/release-20238/#breaking-changes), "MQTT" section "Naming of MQTT entities"
New entity naming [guidelines](https://developers.home-assistant.io/docs/core/entity/#entity-naming)

### Currently we are getting next log lines in the HomeAssistant 2023.8:

2023-08-22 20:31:27.404 WARNING (MainThread) [homeassistant.components.mqtt.mixins] **MQTT entity name starts with the device name in your config** {'unique_id': '3a7c01fb9b5a_output_07', 'name': 'heat1floor output 07', 'availability_topic': 'NetworkModule/heat1floor/availability', 'state_topic': 'NetworkModule/heat1floor/output/07', 'command_topic': 'NetworkModule/heat1floor/output/07/set', 'device': {'identifiers': ['NetworkModule_3a7c01fb9b5a'], 'model': 'HW-584', 'manufacturer': 'NetworkModule', 'name': 'heat1floor', 'sw_version': '20220921 0050', 'connections': []}, 'enabled_by_default': True, 'optimistic': False, 'qos': 0, 'payload_not_available': 'offline', 'retain': False, 'payload_available': 'online', 'encoding': 'utf-8', 'payload_off': 'OFF', 'payload_on': 'ON', 'availability_mode': 'latest'}, **this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'output 07'**

### To fix this we need to strip device name from the beginning of 'name' attribute.
'name': 'heat1floor output 07'
should become
'name': 'output 07'

That will fix issue.

nielsonm236, 2023-08-22:
OK - Should be easy to do. How soon is this needed?

yozik04, 2023-08-22:
Till February 2024 😄

nielsonm236, 2023-09-25:
Just to clarify the above ONLY affects the Auto Discovery messages, but not any of the other Subscribe or Publish messages, correct?

nielsonm236, 2023-10-15:
Addressed in release 20231009 1022

closed 2023-10-15 by nielsonm236

yozik04, 2023-10-21:
I can confirm that warnings are now gone.

nielsonm236, 2023-10-21:
Thank you. Nice when a fix actually works. :-)