nielsonm236/NetMod-ServerApp #201: Output pin change batch REST commands /50 and /51 broken in 20230801 release

peoyli, 2023-08-25:
All outputs go active when using the /50 and /51 commands in the current release (browser only firmware):
http://192.168.1.4:8080/51FFFF0003
returns
1111111111111111
(or "----111111111111"  in my case with the last four outputs set as disabled)

Other examples when run with my test program:
```
   Group          Members          Reverse       Zeropadded
                                   Command    Status/Result
       1 1100000000000000 0000000000000011 0000000000000011
                                51FFFF0003 ----111111111111
       2 0110000000000000 0000000000000110 0000000000000110
                                51FFFF0006 ----111111111111
       3 0011000000000000 0000000000001100 0000000000001100
                                51FFFF000c ----111111111111
       4 0000100000000000 0000000000010000 0000000000010000
                                51FFFF0010 ----111111111111
       5 0000010000000000 0000000000100000 0000000000100000
                                51FFFF0020 ----111111111111
       6 0000001000000000 0000000001000000 0000000001000000
                                51FFFF0040 ----111111111111
       7 0000000100000000 0000000010000000 0000000010000000
                                51FFFF0080 ----111111111111
       8 0000000010000000 0000000100000000 0000000100000000
                                51FFFF0100 ----111111111111
       9 1111000000000000 0000000000001111 0000000000001111
                                51FFFF000f ----111111111111
      10 0001111000000000 0000000001111000 0000000001111000
                                51FFFF0078 ----111111111111

```

nielsonm236, 2023-08-25:
Yikes! Thanks for reporting. I will get right on it.

nielsonm236, 2023-09-25:
Found the source of this bug. It will be fixed in the next General Release.

nielsonm236, 2023-10-15:
Addressed in release 20231009 1022

closed 2023-10-15 by nielsonm236