nielsonm236/NetMod-ServerApp #202: Documentation: about 16 channel relay modules (starting on page 118)

peoyli, 2023-08-26:
I have mentioned this (somewhere, 'issue' or email) before, but I actually happened to find a 16 channel module that works with the HW-584 board without any modification, and it's probably one of the cheapest too (500SEK for five (from Amazon), including shipping from Germany, which with the broken Swedish currency is about $9/each).

It seems like the AZDelivery store on amazon.com (https://www.amazon.com/s?me=A28APF1C6NOVWH&marketplaceID=ATVPDKIKX0DER) only include a small part of what's available on their own site (and through Amazon in Europe), but it's visible on their site (even when selecting US for location)
https://www.azdelivery.de/en/products/16-relais-modul?variant=19860880162912

peoyli, 2023-08-26:
I just found out that they changed the design (or found another to send to JLCPCB for manufacturing), so the batch I of them I got for my project have the same issue as described in the manual. Will have to look into it what needs to be modified to make these work (or ask for a batch of their old, directly working design)

peoyli, 2023-08-28:
The 16 channel relay module that works directly is called HW-523 (-A for 12V, and -B for 5V relays). Also printed as identification on some of the boards I found online is "HP20014116A". Another identification string is between the relay rows which says "16 Relay Module Low Level Trigger". There are probably more variants of these, but the important thing to look for is the power LED near +5V on the 20p header. If it's there, there is a high probability that the board works directly with the HW-584 controller.
One of the many sellers of this board design (cheap board, but shipping costs adds up quick for each board added)
https://www.aliexpress.com/item/1005004962029461.html

nielsonm236, 2023-10-06:
Hi - I'm updating documentation now and planning to include your comments above. A couple of questions:
1) When you say it "works directly" you are referring to the IO Pin voltage levels?
2) It looks like there is still the board-to-board interference problem. Do you use a ribbon cable to connect the boards?
3) Can you send a photo of a Network Module connected to one of these boards?
Thanks
Mike

nielsonm236, 2023-10-08:
I'm going to document this as another potential solution to the 16 channel relay board "control pin interface problem", but after my own searches for the HW-523B I was concerned that I would get the right board when ordered. The paragraph above from @peoyli seems to be the best search information, but some vendors are showing a very different board design using the same "HW-523B" name.

The primary advantage of this board is that the control pin voltages and current paths are compatible with the 3.3V logic of the Network Module HW-548 board. I haven't bought one so I haven't reverse engineered the circuit design - but @peoyli reports that it works.

Some additional design information from @peoyli:

If you are going to buy these boards a key seems to be to look for this LED in the lower corner of the 16 Channel relay board. I'm not convinced that the seller's photos will match what they ship to you.
[image]

A couple of photos of a Network Module attached to the board.
[image]
[image]

The above was just for development test, and obviously this isn't the way you would want to mount the boards for actual use. Suggested is to use the following type of connectors soldered to the bottom of the Network Module HW-548 board, or to use a ribbon cable as I've illustrated in the Manual for other 16 channel boards.
[image]

The above can be bought on eBay. Search terms: 2X10 Pin 2.54 mm Stackable 11mm Long Legs Female Header For Arduino Shield

nielsonm236, 2023-10-15:
Addressed in release 20231009 1022

closed 2023-10-15 by nielsonm236

nielsonm236, 2025-02-27:
Feb 27 2025: Adding an update to this closed issue.

While it is reported that this relay board works without modification I have some doubts.

Another user bought a 5V relay version of this board and it did not work properly. While helping that user figure out why it did not work I reviewed this issue and find the same problems on the 12V relay board shown above. If you look at the photo above showing the corner of the board where LED1 is located you can plainly see the 5V trace leading from the header over to resistor packs PR1 and PR2. If 5v is applied to those resistor packs the protection diodes on the HW-584 will be in their active region, which will cause intermittent operation of the relays. This is the same problem seen on most 16 Channel relay board designs.

So, I think modifications are required on these boards for reliable operation. I will update the manual to reflect these findings.