nielsonm236/NetMod-ServerApp #215: OPTION BYTE reading failed

techsd, 2023-11-14:
CONFIGURATION:
Hardware: ST-LINK
Programming mode: SWIM
Device name: STM8S105x6
Port: USB

PROGRAM MEMORY status:
[0x008000 - 0x00FFFF]
No File
Not programmed
Memory checksum: 0x0

DATA MEMORY status:
[0x004000 - 0x0043FF]
No File
Not programmed
Memory checksum: 0x0

OPTION BYTE status: *
No File
Not programmed
Option byte 0: 00
Option byte 1: 00
Option byte 2: 00
Option byte 3: 00
Option byte 4: 00
Option byte 5: 00
Option byte 6: 00
Option byte 7: 00
Option byte 8: 00
Memory checksum: 0x0

[image]

[image]

[image]

[image]

[image]

techsd, 2023-11-14:
only rst and swim are connected 
I tried to turn off the power of the device without pulling out the key, waited until the final power disappeared 
then tried to connect again 
the same thing happened 
and then I tried to connect gnd without turning off the power and without pulling out the switch 
and now it says that the device is protected but the erase option is not active 
> Reading OPTION BYTE area...
Error : The device is protected
Operation aborted
Error : < OPTION BYTE reading failed.

techsd, 2023-11-14:
I tried this - This process ensures a proper reset, which can help in some cases when the microcontroller is in a locked state:
connect the NRST pin to GND, and then turn off the device. Wait a few seconds, turn the device back on, and then disconnect NRST from GND.

but did not help to solve the problem, 
the message 
'> Reading OPTION BYTE area...
Error : The device is protected
Operation aborted
Error : < OPTION BYTE reading failed.'

jmcvieira1, 2023-11-14:
Hello, I will try to help you, from what I understand from the information you published, you are missing the GND and +3.3V connection between the usb programmer and the networkmodule. You can consult all the information in the manual available on the firmware download page, here you have a screenshot of the connections (page 83) you have to follow all the instructions to be able to communicate with the networkmodule, I hope it helps
 Jose Vieira
[image]

techsd, 2023-11-14:
> Hello, I will try to help you, from what I understand from the information you published, you are missing the GND and +3.3V connection between the usb programmer and the networkmodule. You can [...]

no, its ok with gnd and +3,3V. BUTTT!!!
I had to change the dongle and for some reason tugging on the 3.3V issue did not work for a long time, but when I tried it on another laptop and on different usb ports, it worked later

closed 2023-11-14 by techsd