nielsonm236/NetMod-ServerApp #216: programing using SWIM

TheArtizan, 2023-12-11:
First of all thanks for this great project, about 2 years ago I programmed 3 of these modules for my house following your instructions and all is well and they are still working now. 
I needed to use some more modules and so i followed the same programming procedure however I have run into problems and now the yellow LED on the EJ45 socket nologer lights and modules are unsuable. I have tried this on 2 boards and the same result. Would you think it is possible to couple up my old working board and read the program memory then swap out the board for the none working one and program it. 
I am at loss to know how to get the boards running I have tried loading a NetworkModule.stp first  also the CodeUploader.sx followed by Strings.sx followed NetworkModule-MQTT-Home-UPG.sx with the RAM board connected. Many thanks

nielsonm236, 2023-12-11:
Some questions to get you going again:
1) Do you know the revision that was installed on the boards that were working? At some point I started showing the revision at the bottom of the Configuration page.
2) For the boards that are having problems, are these boards that were working and only stopped working when you attempted to install the code?
3) If new boards are having the problems: You don't need the .stp file (follow instructions in the latest Manual). Your process to intall the CodeUploader followed by Strings followed by .sx file is correct. When you installed the CodeUploader did it run? (It must have or you wouldn't have gone on to installing Strings and .sx file ... just making sure I understand when things went South).
Mike

TheArtizan, 2023-12-12:
1, Hello Michael the Code Revision 20230125 2003 MQTT UPG .. is working on the 3 boards that I set up a couple of years ago.
2 The new factory set boards were working until I followed the instructions and tried to upload the new code. I have one board left that I have not programed yet.
3. I installed as per instructions in the manual. I did try the original computer which was my sons and was running widows10 but I appear to be unable to rescue  the boards from what ever I have done.
When I upload I get "Program memory successfully verified" on the STVP

I did run the STVP program this time using a window7 virtual machine on my computer.

The RJ45 socket green LED is on but the yellow one now does not light, they both were illuminated before I uploaded the code.
I have one board left to try which has not had the write protection lifted.
Could it be that I have faulty modules, has this ever happened to you, it would be nice to get to the bottom of it so we can pass on whatever has happened here to others.
Many thanks David

nielsonm236, 2023-12-12:
Hi David,
Re: 1) Is this also the code you tried to install on the new boards? Or was it some other version?
Let's leave the third board alone for the moment and focus on the two that are not working. I really doubt they are defective, and we should be able to get them working. No, I haven't seen this before ... and the couple of times someone was reporting "doesn't connect" eventually it turned out to be some mis-step in programming or wiring to the network.

Next questions:
When you used STVP to install code on the problem boards: If it is not the same version as on your working boards what version did you use? I could re-check if there is a problem in that version.  Also, down in the release history 20230125 2003 is still available and can be downloaded.
When you used STVP to install code on the problem boards you were just installing the Code Uploader, right? (then if it would run you would use it to install strings and the targeted firmware).

Assuming the above doesn't tell us anything, can you verify that the Option Byte tab shows "Read Out Protection OFF" like this:
[image]

If that looks OK I wonder if we have initial settings wrong or a conflict on the network. The default settings are 192.168.1.4, Port 8080. Is your network set up for IP addresses 192.168.1.xxx? Any chance there is a conflict with 192.168.1.4? These question may be a rathole because I think the RJ45 LEDs would still show activity.

TheArtizan, 2023-12-13:
Hello Michael
I have managed to get the boards working, I had success by coupling the RAM and taking off the write protection then uploading ONLY the Code Uploader.
Then because I am on a Mac I had to go to System Preferences/Network and then use settings like on the screen shot and enter the address of the ROUTER and not the IP address, which is how I read the manual. After plugging the Ethernet cable directly into the computer, I was able to go to 192.168.1.4:8080/72. on the browser for the Code Uploader page. Then re-   upload the NetworkModule-CodeUploader.sx then upload the NetworkModule-Strings.sx and then NetworkModule-MQTT-Home-BME280-UPG.sx(Code Revision 20231009 1022 MQTT Home BME UPG), and now all three boards are working.

I was originally loading the CodeUploader, Strings and MQTT-Home-BME280-UPG via the STVP, something corrupted the data, at one point after I had loaded the firmware when I rebooted the module and tried to read the current tab from the STVP and the data showed FFFFFF in all sectors, when I tried to view via a browser I just got a line of  ????????????

I also coupled up the PCF8574 expander and while I can see a Configuration tab on the browser nothing happens when I click it I have checked the wiring and SDA, SCL share the same pins as the RAM module I am taking the VCC from the 3.3V the same as the RAM.

Thank you so much for you efforts with this project,
All the best,
David - from rainy UK

image.png (view on web)<https://github.com/nielsonm236/NetMod-ServerApp/assets/63531544/7f5902a4-1824-4bf7-a956-f31b027288d4>

nielsonm236, 2023-12-13:
I should go review the Manual again and see if I can improve it. Perhaps it is not clear enough.

First a jargon correction. You mention attaching "the RAM". That is really I2C EEPROM so that it remains non-volatile through power cycles.

Next: Yes, it you are going to use Upgradeable (UPG) builds you must first use STVP to load the Code Uploader. When the Code Uploader runs it will copy itself to I2C EEPROM so that it can be automatically accessed in the future without STVP. Once the Code Uploader is installed and running you can use it to upload the Strings file and the target runtime firmware (like NetworkModule-MQTT-UPG.sx) without STVP. So, once you've gone through this process the first time you shouldn't have to use STVP again (the exception being if you inadvertently upload non-UPG code ... which has no means of accessing the Code Uploader ... and you get to go back to STVP to correct the mistake).

So if you used STVP to program the Strings.sx file that is the source of your corruption. The Strings.sx file is not firmware, it is just a set of strings that contain the Webpages. Those are strored in I2C EEPROM and can only be put in I2C EEPROM by the Code Uploader. Placing those strings in I2C EEPROM is how I freed up enough space in the STM8 Flash for additional features (including the ability to upload code releaser over Ethernet!)

But I think clarification in the manual may be needed with regard to "first time programming of a module with UPG firmware". The key question to answer is "OK, I installed the Code Uploader using STVP ... but how do I access it if the default IP Address is not compatible with my network????"  Yeah, you can use direct connect, but I find that to be rather clunky, as you did. Maybe I can come up with some simplified instructions for putting in an IP address with STVP. Believe it or not you CAN do that, but it requires manually editing the "Data Memory" (the STM8 EEPROM). Not rocket science, but a step beyond what most users are used to dealing with, and you can easily corrupt the STM8 EEPROM ... causing further confusion.

With regard to the PCF8574: Look near the bottom of the Configuration page. You'll see something that looks like "PCF8574 0" or "PCF8574 1". If "0" it means the firmware did not detect a PCF8574. That probably indicates a wiring error when you attached the PCF8574.  If "1" the PCF8574 was detected and the menu items should work for setting it up. FYI, there is no standard for the order of connections on I2C modules, so it is easy to inadvertently swap the Clock and Data connections.

Let me know if you have issues or suggestions. Also feel free to email me at nielsonm.projects@gmail.com
Mike

closed 2023-12-20 by nielsonm236