nielsonm236/NetMod-ServerApp #221: changing settings

tomaszvil, 2024-03-02:
Hello
I am using your firmware to Network Module  Web_Relay_Con V2.0 HW-584, I am interested in your project.

	After installing several HW-584 modules in the same network, I have a problem with them .
The HW-584 network modules change their configurations on their own without my intervention, change are mac adrr, name,  IO Type and Invert Boot State, are changed to incorrect ones .
Button refresh and reboot do not restore the settings, only disconnecting the power causes a proper start to restore the settings.
It also happens that it stops working completely, there is no communication and only restoring to the initial settings (5s reset) revives the module. 
When I read data from processor  (settings) area of the processor, incredibly random data is stored in the eeprom.

The hw 584 network modules are connected via the MQTT protocol with Home Assistant.
Only Home Assistant controls my network modules.
I use last version NetworkModule-MQTT-Home.
I bought several modules from different sellers, all of them had the same problem.
Each network module is connected to its 16 Relay Module Low Lewel Trigger.
When I used one HW-584 network module , the settings changed once in 1 months.
In my project I need to use 5 modules in different locations on the same network.
When there were already 4 HW-584 network modules in my network , the settings changed randomly in different modules, very often even twice in 3 days.
Currently, I have 2 active HW-584 network modules and I have noticed that the settings change less frequently, about once a two week, sometimes more often.
I have set an individual IP address and an individual MAC address for each HW-584 network device.
There are a dozen or so computers and about twenty IP cameras in the network
I appreciate your dedication and contribution to this project and the very detailed documentation .
Please help me solve my problem.
Sorry for my English.

Tomasz
[image]
[image]

nielsonm236, 2024-03-02:
It isn't the HW-584 hardware, or the router. I think you are OK there. This must be a firmware bug.

I haven't seen any other reports of this problem, so we need to dig a little deeper.
1) At the bottom of the Configuration screen the firmware reports the Code Revision. Can you send a screen shot of that?
2) Are all the HW-584 modules that exhibit the problem at the same Code Revision?
3) Do you have a feel for how often the relays change state? In particular, do you have any situations where most of the relays (or all 16) are commanded to change state at the same time?
4) When you said "Button refresh and reboot do not restore the settings, only disconnecting the power causes a proper start to restore the settings" do you mean that the settings are returned to being correct just by disconnecting and reconnecting power? Or are they still incorrect?

My initial questions above are to try to understand the following:
a) If the problem is specific to an older code revision.
b) If the problem is related to hardware reset that might be caused by power glitches with many relay's switching state all at once.

Note that I have seen similar corruption of the EEPROM but only during development of code. Typically it is because I have an errant pointer that overwrites the EEPROM. I've built in a lot of code protection to prevent this, To my knowledge it hasn't occurred in the field - only in my test configurations while code is being changed. But there is always a first time :-)

OTHER USERS: PLEASE RESPOND IF YOU HAVE SEEN THIS.

It is not certain what will happen if the HW-584 modules have the hardware reset triggered frequently. That is very difficult to test, and I know that it is much more difficult to control DC power glitches when more relays are attached. There are several people using 16-relay boards successfully, so we just need to try to understand what might be different in your situation.
Mike

tomaszvil, 2024-03-02:
Hello Mike
I am very pleased that you  so interested in my problem and you trying to help me.
1. I am attaching the entire screenshot of the code version, I am using the latest available version. (screenshot)
2. All HW-584 modules have the same, latest firmware and all of them have the problem..

3. There are not many relay status changes, several times a day, several relays, maximum 4-5 at the same time, are switched on. At the same time, it turns the on or off. maximum 3 relays
      There is no rule when the HW-584 module hangs, the home assistant reports a loss of communication at different times, certainly not when the relay status changes, and sometimes loss communication when they are all turned off, there is no situation where many relays are turned on at the same time.
4. Sometimes disconnecting and turning on the power restores the settings, but it often happens that a lot of errors are saved in the eeprom memory, along with changing the MAC address or IP address, only a 5s reset restores the factory settings.
5. In my solutions, I used good quality Meanweel 12v power supplies to power the relays block and I take 5v from the relays block board, photo from voltage measurement.
Tomasz from Poland
[image]
[image]

nielsonm236, 2024-03-02:
This is very confusing. Everything you are doing looks right.
I'm going to take my test configurations (9 HW-584 modules) back to that revision level using Home Assistant to see if I can replicate this. I don't have relays connected - just LEDs to indicate IO status.
About how long do the modules run before you see a failure?

tomaszvil, 2024-03-03:
Hello Mike
I am very impressed with how seriously you took the topic.
When I had four HW-584 modules configured, they worked correctly for a maximum of a few days.
Now I have two modules in use, they can correctly work for 2-3 weeks, but sometimes after 2-3 days one or the other will crash.
But I will add once again that I have a quite heavily loaded network of IP cameras and dozen computers. 
Although yesterday it stopped at night when the network traffic was low, and the home assistant did not switching anything.
Please advise me what else I could check. although I must say that I have been struggling with this problem for a long time, when I exhausted all my ideas, I decided to write to you
Kind regards Tomasz

nielsonm236, 2024-03-03:
I just don't see how heavy network loading could cause a problem. If there is a very large amount of messaging through the MQTT Server/Broker it is possible that messages might get lost I suppose. But I don't see how this would cause corruption of the EEPROM in the HW-584. So I will try to get a test going within the next day.
If I can get a failure to occur I might then be able to track it down.
Mike

nielsonm236, 2024-03-03:
I've configured 5 HW-584 devices to start running a Home Assistant test tonight. I'll add more if I don't see any replication. These first 5 devices run the "Upgradeable" version of code. Operationally this should be identical to the non-Upgradeable code, but it allows me to change code without needing to use the SWIM interface.
As FYI I've configured Home Assistant to turn on Output 1 at 5 minutes past the hour on all devices, and to turn those outputs off at 15 minutes past the hour. This is just to see what happens. I'll increase the complexity if I don't see any replication.
Also as FYI my network is fairly busy and includes 16 security cameras. Also a teenager playing video games :-)
I will keep you posted.
Mike

nielsonm236, 2024-03-03:
While this first test runs I thought of a possible issue.
During some recent code development (not released yet) I found that the MQTT Username and Password fields might allow entry of forbidden characters. If you only use the characters that the pop-up says you can use (hover over the fields to see the pop-up) then you should be OK.

tomaszvil, 2024-03-03:
Hi Mike
I used only allowed characters to save the password and name.
Tomasz

nielsonm236, 2024-03-03:
Tomasz - Please email me at nielsonm.projects@gmail.com
I want to be able to exchange some information about our networks and network activity that you might not want visible to the public. I'm just thinking ahead that if I can't replicate the problem we need to look deeper into what is different between your network  and Home Assistant configuration and utilization that is different from mine.
Right now I think you have found some corner case in the firmware that is causing a pointer over-run when writing EEPROM. The trick will be figuring out what particular event is making this happen.
After we figure this out we can come back here and publish only the facts that might be useful to another user.
Mike

nielsonm236, 2024-06-15:
Despite a significant number of email exchanges with @tomaszvil we've been unable to resolve the issue. I believe this to be a power glitch problem, but @tomaszvil has performed a significant number of measurements, power rearrangements, and tests and the problem persists. I am unable to reproduce in my setups, with the exception of seeing a similar problem with a degraded power supply long ago (issue #125 in Closed Issues).

I'm out of ideas.

yozik04, 2024-07-23:
@nielsonm236 Hello Michael!
Couple days ago I've noticed a similar problem with one of the modules. I use just two at the moment for house heating and water circulation. Power to the module is fine. 5.23V almost constant. I do control just one of the relays with it during this time of the year.
The device does not crash, it still responds to port 80 but HomeAssistant tells that device became unavailable. I assume MQTT part has stopped working. Configuration page shows some junk symbols in MAC Address and I see some value changes in IO table that I did not do.
I tried to change MAC Address and fix IO table. IO was fixed but MAC Address is corrupt the same way. Interesting that even if I reboot then it starts to work for around a day then it disconnects from MQTT again.

MQTT Explorer:
[image]

Configuration after I fixed IO Table:
[image]

yozik04, 2024-07-23:
and IOControl page screenshot:
[image]

Note the difference between what is in MQTT Explorer and on this screenshot.

yozik04, 2024-07-23:
I can try to install 20240612 to check if it will fix the problem. I can dump EEPROM Data block before that if needed.

yozik04, 2024-07-23:
BTW NetModule calculates page length wrongly.

```
curl -v http://192.168.5.18/61                                                                               19:04:32
*   Trying 192.168.5.18:80...
* Connected to 192.168.5.18 (192.168.5.18) port 80
> GET /61 HTTP/1.1
> Host: 192.168.5.18
> User-Agent: curl/8.6.0
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length:04738
< Cache-Control: no-cache, no-store
< Content-Type: text/html; charset=utf-8
< Connection:close
<
<html><head><link rel='icon' href='data:,'><meta name='viewport' content='width=device-width'><style>.s0{background:red;}.s1{background:green;}table{border-spacing:8px2px}.t1{width:100px;}.t2{width:450px;}.t3{width:30px;}.t8{width:60px;}.c{text-align:center;}.ip input{width:27px;}.s div{width:13px;height:13px;display:inline-block;}.hs{height:9px;}</style><title>heat1floor: Configuration</title></head><body><h1>Configuration</h1><form onsubmit='return m.s(event);return false'><table><tr><td>Name</td><td><input name='a00' value='heat1floor' pattern='[0-9a-zA-Z-_*.]{1,19}' required title='1 to 19 letters, numbers, and -_*. no spaces'/></td></tr><tr class='hs'/><tr><td>IP Address</td><td><input name='b00' class='ip'/></td></tr><tr><td>Gateway</td><td><input name='b04' class='ip'/></td></tr><tr><td>Netmask</td><td><input name='b08' class='ip'/></td></tr><tr><td>Port</td><td><input name='c00' class='t8 port'></td></tr><tr><td>MAC Address</td><td><input name='d00' required pattern='([0-9a-fA-F]{2}[:-]?){5}([0-9a-fA-F]{2})' title='aa:bb:cc:dd:ee:ff format'/></td></tr><tr><td>Features</td><td class='f'></td></tr><tr class='hs'/><tr><td>MQTT Server</td><td><input name='b12' class='ip'/></td></tr><tr><td>MQTT Port</td><td><input name='c01' class='t8 port'></td></tr><tr><td>MQTT Username</td><td class='up'><input name='l00' value='heat2floor'></td></tr><tr><td>MQTT Password</td><td class='up'><input name='m00' value='JCYHxeWp24'></td></tr><tr class='hs'/><tr><td>MQTT Status</td><td class='s'><div class='s1'></div> <div class='s1'></div> <div class='s1'></div> <div class='s1'></div> <div class='s1'></div></td></tr><tr class='hs'/></table><table><tr><th>IO</th><th>Type</th><th>Invert</th><th>Boot state</th></tr><script>const m=(e=>{let t=['b00','b04','b08','b12'],$=['c00','c01'],r={'Full Duplex':1,'HA Auto':6,MQTT:4,DS18B20:8,BME280:32,'Disable Cfg Button':16},n={disabled:0,input:1,output:3,linked:2},o={retain:8,on:16,off:0},l=document,a=location,p=l.querySelector.bind(l),i=p('form'),c=Object.entries,d=parseInt,_=e=>l.write(e),s=(e,t)=>d(e).toString(16).padStart(t,'0'),u=e=>e.map(e=>s(e,2)).join(''),f=e=>e.match(/.{2}/g).map(e=>d(e,16)),b=e=>encodeURIComponent(e),h=e=>p(`input[name=${e}]`),g=(e,t)=>h(e).value=t,x=(e,t)=>{for(let $ of l.querySelectorAll(e))t($)},E=(e,t)=>{for(let[$,r]of c(t))e.setAttribute($,r)},y=(e,t)=>c(e).map(e=>`<option value=${e[1]} ${e[1]==t?'selected':''}>${e[0]}</option>`).join(''),A=(e,t,$,r='')=>`<input type='checkbox' name='${e}' value=${t} ${($&t)==t?'checked':''}>${r}`,S=()=>{let r=new FormData(i),n=e=>r.getAll(e).map(e=>d(e)).reduce((e,t)=>e|t,0);return t.forEach(e=>r.set(e,u(r.get(e).split('.')))),$.forEach(e=>r.set(e,s(r.get(e),4))),r.set('d00',r.get('d00').toLowerCase().replace(/[:-]/g,'')),r.set('h00',u(f(e.h00).map((e,t)=>{let $='p'+t,o=n($);return r.delete($),o}))),r.set('g00',u([n('g00')])),r},T=(e,t,$)=>{let r=new XMLHttpRequest;r.open(e,t,!1),r.send($)},j=()=>a.href='/60',k=()=>a.href='/63',v=()=>a.href='/61',w=()=>{l.body.innerText='Wait 5s...',setTimeout(v,5e3)},B=()=>{T('GET','/91'),w()},D=e=>{e.preventDefault();let t=Array.from(S().entries(),([e,t])=>`${b(e)}=${b(t)}`).join('&');T('POST','/',t+'&z00=0'),w()},q=f(e.g00)[0],z={required:!0};return x('.ip',e=>{E(e,{...z,title:'x.x.x.x format',pattern:'((25[0-5]|(2[0-4]|1[0-9]|[1-9]|)[0-9])([.](?!$)|$)){4}'})}),x('.port',e=>{E(e,{...z,type:'number',min:10,max:65535})}),x('.up input',e=>{E(e,{title:'0 to 10 letters, numbers, and -_*. no spaces. Blank for no entry.',maxlength:10,pattern:'[0-9a-zA-Z-_*.]{0,10}$'})}),t.forEach(t=>g(t,f(e[t]).join('.'))),$.forEach(t=>g(t,d(e[t],16))),g('d00',e.d00.replace(/[0-9a-z]{2}(?!$)/g,'$&:')),f(e.h00).forEach((e,t)=>{let $=(3&e)!=0?A('p'+t,4,e):'',r=(3&e)==3||(3&e)==2&&t>7?`<select name='p${t}'>${y(o,24&e)}</select>`:'';_(`<tr><td>#${t+1}</td><td><select name='p${t}'>${y(n,3&e)}</select></td><td>${$}</td><td>${r}</td></tr>`)}),p('.f').innerHTML=Array.from(c(r),([e,t])=>A('g00',t,q,e)).join('</br>'),{r:B,s:D,l:v,i:j,p:k}})({b00:'c0a80512',b04:'c0a80501',b08:'ffffff00',c00:'00050',d00:'��',b12:'c0a80505',c01:'0* transfer closed with 8 bytes remaining to read
* Closing connection
curl: (18) transfer closed with 8 bytes remaining to read
075b',h00:'00000303030303008880800000008000',g00:'0f'});</script></table><p><button type=submit>Save</button> <button type=reset onclick='m.l()'>Undo All</button></p></form><p>Pinout Option 1<br/>Code Revision 20231009 1022 MQTT Home ............<br/><a href='https://github.com/nielsonm236/NetMod-ServerApp/wiki' target='_blank'>Help Wiki</a></p><button title='Save first!' onclick='m.r()'>Reboot</button><br><br><button title='Save first!' onclick='m.l()'>Refresh</button><br><br><button title='Save first!' onclick='m.i()'>IO Control</button> <button title='Save first!' onclick='m.p()'>PCF8574 Configuration</button></body></html>%
```

yozik04, 2024-07-23:
One more thing. If I click "Reboot" button, after intermediate "wait 5 seconds" page Network module still does not communicate via MQTT. Only power circle fixes the problem.

nielsonm236, 2024-07-24:
The information that some pages may not be properly sized may be a clue. I will start there. If a page is not sized properly it might cause a buffer over-run and that could lead to data corruption. Very small chance that is the problem but it needs to be corrected.

If the MAC address is corrupted it will interfere with HA. The MAC address is used as part of the device ID's.

I'm not sure how the Browser continues to work with a corrupted MAC address. The router should be very unhappy about that and shouldn't be able to find the device. But ... I've forgotten more than I remember so maybe it can still work if the IP address is unchanged.

Mike

yozik04, 2024-07-24:
I do not think that page size is the problem. It becomes a problem only when MAC Address get's corrupt. Before that page sizes are correct.

I see when corrupt it sends next data for MAC Address:
```hexdump
0000000 bfef efbd bdbf
0000006
```
Which is 6 bytes.

But when unit works correctly it sends:
```
0000000 6133 6337 3130 6266 6239 6135
000000c
```
Which is 12 bytes.

nielsonm236, 2024-07-24:
I think the hexdump might not be correct. A MAC address without a mix of digit and alpha characters is very unlikely.

Still the question is "how did the MAC corruption occur", because your screen captures very clearly show that did happen. Since your system operated a very long time I agree power glitching is unlikely. Now I'm thinking tha tis also true for @tomaszvil. A buffer over-run can corrupt just about anything, depending on where the compiler placed those things in RAM. So I agree it is a long shot. 
Interesting that the page size is valid until the corruption occurs. The corruption of the MAC address might cause this if the resulting MAC content is a not-allowed "character" in HTML. I will still transmit all 12 bytes of the MAC, but the receiving Browser will toss not-allowed bytes and not count them in the received page size (just as it does with extra spaces). At least I think that is what happens. This is going to be very difficult to find.
I will have to go back and look at what I changed in the 20231009 release. To your knowledge you didn't have any problems prior to that release?
Some things that might cause problems that could explain why I don't see it, but you and @tomaszvil do see it:
- The Configuration itself, in particular with use of MQTT Username and Password. I don't use Username and Password.
- I recall fixing the filtering around Device Name, MQTT Username, and MQTT Password.
- I have "barely tested" code for a Software Defined Radio project and for Browser Login ID. Both requestors evaporated before helping with test. All that code is fenced with #if statements, but it is possible some change did not get fenced and is causing a problem. Or, some allocation of resource shifted RAM locations about and exposed a latent problem.

First I need to see if I can get this to reproduce so that I can add some debug. We may have to exchange some information by private message so I can set up a duplication of your Configuration in greater detail.

Mike

yozik04, 2024-07-24:
Regarding "Browser will toss not-allowed bytes". This can't be true as I was using `curl` command line tool to capture what HTML I am getting and verify if packets received are full. Curl clearly tells that received byte count is different from what was sent in ContentLength header. Next time I will pipe curl output to hexdump to see what exactly is there. Maybe will use wireshark to capture TCP packets as well.
My module worked on 20231009 firmware since it's release. So it is at least half a year without any issues. Suddenly I got this corruption three times in a period of couple days. After power circle it worked for some hours (4-6) and got corrupted again. After Monday's power circle and settings update It still works without issues (23 hours has passed).
I do not think that the problem was introduced in 20231009 because I've seen some corruption of settings using previous releases as well. No idea what can be the cause.

nielsonm236, 2024-07-24:
OK. Now I'm completely baffled again. It ran fine for months then started to repeatedly have a EEPROM corruption problem? Thinking out loud to see if it provokes any ideas:
1) Could be some aliasing issue causing a buffer over-run. But this should be completely random and won't just start occurring repeatedly after months of no issue.
2) Maybe some network activity issue that wasn't there before. I'm aware that the hardware filtering in the ENC28J60 still lets a lot of network probing activity through. Firmware then filters out the bogus activity that the ENC29J60 did not filter. Could there be too much bogus activity? Perhaps. But firmware should just start dropping messages if overrun. It may be that there is a leak in that process that causes a buffer over-run. As an FYI, the "bogus activity" I've seen seems related to Microsoft Windows probing the network for resources, and Smart devices and Alexa dots probing the network for devices. And other probing activity that I think is coming through from the Internet. Sometimes there is just a huge storm of this going on.
3) There aren't any cumulative processes going on in the firmware. I could search to re-verify that there aren't any counters that can roll over and cause an issue. But even if there is one, it should cause a problem ONCE, then after a reboot it would be many months again before the problem resurfaces. So, that can't be it.
3a) Well, OK the Link Error Statistics output accessible via the /66 command has two counters. They should just roll over at max count. Might be useful to see if anything is showing up there. Like a Stack Overflow.
4) There is always the power noise issue ... that was actually a very repeatable source of corruption that I had in my own test bed. Cheap power supplies can slowly go bad over a long time (which is the problem I had).
5) Is it possible something is repeatedly writing the EEPROM and wearing it out? I've been very careful to avoid that. My test bed has far exceeded the specs for EEPROM and Flash writes, and still I don't see the corruption issues. And if this is a wearout problem, you've seen corruption in the past but then the problem goes away. That's not EEPROM wearout.
6) Over-heating of the STM8 processor might cause processor timing issues and EEPROM/Flash write issues. And those issues could be transient. I have been suspicious of this because of the fact that the STM8 is 3.3V, and users regularly want to hook up 5V logic. If done correctly it is no problem. If done incorrectly it will cause thermal issues, and at worst device failure.
7) Is it possible some part of the Flash has become corrupted, and reloading the firmware will correct it? Seems unlikely Flash could have any corruption but the code still runs. The Link Error Statistics /66 page includes an ILLOPF counter to expose hardware detection of an illegal opcode. Might be a clue.

If we can think of any instrumentation to confirm or reject a suspect issue let's do it while you are seeing the problem.

Mike

nielsonm236, 2024-07-24:
Regarding ""Browser will toss not-allowed bytes" I'm running on recollection here which could be faulty. I know that if I code a string with multiple spaces in a series it will appear at the Browser as a single space. I can't remember where the spaces are concatenated .... perhaps just in the Browser itself. I also vaguely recall that if I sent bytes outside of the Browser character set they can cause problems. But I don't know if that will affect receive count or not.

Notes on MAC adddress storage:
The MAC address is stored as 6 uint8_ bytes in the EEPROM. This is used in most processes throughout firmware, and is also stored in the ENJ29J60. But the MAC is also stored as 12 hex characters in a string in RAM for use in display functions and in generating MQTT IDs. This could explain why some connections persist (those using the uint8_t bytes), and some fail (those, like MQTT, using the MAC string).
So it is possible the EEPROM is not being corrupted, but the MAC string IS being corrupted in RAM.
Again, thinking out loud.
Mike

yozik04, 2024-07-25:
It is a great list of potential places for looking.
2. Network activity: can be. I have many different devices but no new ones. Actually couple of Android phones I gave to kids with custom firmwares... But let's see if the issue will reappear.
3a. I will get statistics when/if it will crash again. Interesting to see what is there when it is with corrupt MacAddress.
4. Will try a different higher quality power adapter after next crash. That was my initial assumption but when I measured the voltage on that I decided to keep it. It produces sound but still works.
5. I intentionally do not use retain flag.
6. It is in a closed box with some holes for cables only. Will measure the temperature in the evening.

Regarding MacAddress storage. I also think that MAC string IS being corrupted in RAM. Because when I power circled last time the unit MacAddress reappeared in the Configuration page intact.

Something different happened this time. I will create a separate issue not to pollute this one. #228 
State in MQTT Explorer is equal to what I see in IO Control page.
I do not see corruption in MacAddress this time. MQTT Broker still receives temperature data. Configuration page does not show any corruption. Content-Length equals with the size received.

So this is not the crash we are looking for...

From the page /66:
3a) Link Error Statistics:
31 0000139361
32 0000021198
33 ff5c864594
34 0180000000
35 0000000001

I clicked reboot and `availability` has restored to `online` again.

Nothing to do, let's wait if it will eventually crash.

nielsonm236, 2024-07-25:
My interpretation of the Link Error Statistics report:
Line 31 is "Seconds Since Reboot", so this device has been running over 38 hours. The counter should not roll over for 317 years.
Line 32 is the "Number of ethernet transmissions" from the device since power on or reboot. Same 10 digit rollover count.
Line 33 is very concerning.
- The first two bytes should always be zero. I will look into how they could possibly have any other content.
- The byte that says "86": "8" indicates a Stack Overflow occurred at some time. "6" is the ENC28J60 revision and is correct. The Stack Overflow indicator is retained through power cycles so it could have occurred long ago. It can be cleared with URL /67.
- The "45" is the transmit error counter. This seems high. This is cleared at power on / reboot so this is very unusual.
- The "94" is the receive error counter. This seems high. This is cleared at power on / reboot so this is very unusual.
- The error counts are not necessarily TOO high. I treat them more as a symptom than a clear problem. You can clear them with /67 then check with /66 to see if they are incrementing slowly (probably not a problem) or very fast (probably a real problem). One every few minutes is probably not a problem.
Line 34:
- The "01" suggests an unexpected power loss but might have occurred long ago. I don't think the detector is very reliable.
- The "80" is the number of times the device has been reprogrammed via the SWIM interface.
- The rest of the line indicate there have been no Illegal Opcodes or Watchdog timeouts. This tells me the Flash is not corrupted.
- The counters in Line 34 are cumulative and are retained through a power cycle / reboot. They can be cleared with URL /67.
Line 35:
- Indicates no MQTT errors, but the Broker sent a disconnect message one time since power on / reboot.
- Line 35 counters are cleared at power cycle / reboot or with URL /67.

To get a better view of "what has happened since we started testing" I'd suggest using command /67 to zero out the counters. That way we will eliminate the "long ago" history and see what is affecting operation right now. I recall you are using a Cisco business class switch, which is why you have Full Duplex enabled. I also recall we would see somewhat higher transmit and receive error counts with that switch, but what is shown above is much higher than I would expect even with the Cisco switch.

Mike

yozik04, 2024-07-25:
Uhh, so professional. You are developing like for satellites to launch into the space =)

No, I am not using a Cisco Business class switch. NetModule is connected to some cheap TP-Link TL-SG105E Gigabit Switch which is connected through another switch to the main router. I can disable full duplex if needed, this should work in both modes, I think.

Some years ago one power block has already died with that module, maybe this is why we see unexpected power loss.

Both modules I have are configured to Full Duplex mode, both have DS18B20 temperature sensors attached (first one has 4, second one has just 2). Settings are the same only number of configured IO output's differ.

Before writing here I used "Oxide clean & Protect liquid" to make sure ethernet cable connectors have good contact. But if you say that tx/rx error rates are reset after every reboot then errors before cleaning should not be included.

I have drilled some holes in the box to let some fresh air in. But both chips on the board were not hot to touch. My infrared temperature sensor showed ~50 degrees celsius on the main chip. Anyway, will be cooler now.

Here are both /66 pages of my both modules before reset if we want to compare.
Module 1:
```
Link Error Statistics
31 0000024920
32 0000004033
33 ff5c864594
34 0180000000
35 0000000000
```

Module 2
```
Link Error Statistics
31 0000752880
32 0000064662
33 0000860000
34 0000000000
35 0000000000
```

Now I cleared counters on both devices and rebooted them both.

Initial states are almost equal (only 32 differ a bit):
```
31 0000000149
32 0000000050
33 0000060000
34 0000000000
35 0000000000
```

Will recheck what are error rates tomorrow. 20 minutes has passed and still zero.

nielsonm236, 2024-07-25:
Full Duplex shouldn't be used except with those Cisco business class switches.

The ENC28J60 isn't supposed to work very well in Full Duplex. It has bugs in the auto-negotiate hardware, so the specs recommend half-duplex. The only reason Full Duplex was allowed in my firmware is that is seemed to work a little better with the Cisco switch IF the user also forced the port on the Cisco switch to Full Duplex (no auto negotiate). So, having it set to Full Duplex is probably causing some of the network errors, particularly when network collisions occur (which is really a common thing on busy networks).

At the present time I have no space launches planned. ;-)

Mike

yozik04, 2024-07-25:
Uhh, I should have studied the manual more diligently. I have disabled Full Duplex on both modules now, restarted and cleared the counters one more time. Will see how it will go. Thank you for all the help.

jmcvieira1, 2024-07-25:
Hello guys
I'll contribute my two cents, as I already had exactly the same EEPROM corruption problem with one of the modules on my test bench, and it was a bad contact on one of the power wires.
I have a module with the same firmware running outside my house inside a PVC box in the sun. This module only controls an electrovalve and monitors the status of the water well motor, as well as the temperature inside the box.
Here are the errors:
Link Error Statistics
31 0007736224
32 0000427905
33 0000860000
34 0000000100
35 0000000000

Thinking out loud:
I see this as being a hardware/power issue.
If the module worked well for months, what has changed now to start showing these faults?
Did the temperature increase?
Has some of the hardware degraded over time?
If the firmware is the same, shouldn't the same symptom have already manifested? 
Or would there not be other modules showing the same symptom?
What is so special about this module that it has only now presented this error?
If it's not a power supply problem, could it be an electromagnetic interference problem?

J. Vieira

nielsonm236, 2024-07-25:
I hope either switching to Half Duplex or changing the power module gets this back in shape. Otherwise this will be a bugger to figure out. It is interesting that it is so similar to the @tomaszvil report. But he worked very hard on trying to get it working and had no success. I would love to have been at his location to help him track it down. But ... he is 5000 miles away.

The problem with power is that it is not just average voltage level like you see with a meter, but high-frequency noise spikes that you could only see with a high-bandwidth scope, and then only if you get lucky enough to catch it. I found a very stable Lambda industrial power supply to run my test bed. But I still use cheap wall wart power supplies for devices scattered around the house.

nielsonm236, 2024-07-25:
Jose - Your thinking is the same as mine. But I've created many bugs over time, so I always want to think it through. You, Jevgeni, and Carlos have always been very good at trouble shooting issues and making suggestions, so I pay close attention when you bring something up.
Mike

jmcvieira1, 2024-07-25:
In my work I encounter this many times, a power supply that appears to be good but due to one or another factor, whether the temperature or an increase in the network voltage, for example due to the use of photovoltaic panels, and the power supply glitches, these are The most difficult faults to detect as sometimes they only appear at the customer's home due to local conditions.

I have almost all my low voltage devices connected to a centralized ATX power supply, everything that works with 5v or 12v is connected to this source, it included all switches, ip cameras, wireless access points and the network modules.

J. Vieira

nielsonm236, 2024-07-25:
The downside to a centralized power supply that distributes DC voltage is that if you want to disconnect and reconnect only one device you must do that on the DC side of the supply. This is a very noisy power disconnect / reconnect due to arcing. The STM8 reset detector (and similar detectors in most circuits) assumes a power ramp that is much slower than DC switch contact or pushing a wire into a connector. You get a good ramp when you disconnect the AC side of a supply so that the DC bulk capacitance bleeds down the voltage.

If you use DC distribution you can get a similar effect at each of the devices by placing bulk capacitance at the device, but you also need high frequency caps (or an inductor) along with the bulk capacitance to get rid of the high frequency noise created by the disconnect/reconnect arcing.

I haven't experimented with this very much, but I imagine the use of Power Over Ethernet presents a similar problem when Ethernet cables are disconnected / reconnected. I did some testing a couple of years ago with POE and it worked fine. My assumption is that the 48V to 5V converter module had sufficient capacitance to make a reasonable power ramp.

Mike

jmcvieira1, 2024-07-25:
I have several DC circuit breakers in different sectors so I can turn them off and on if necessary, but it's a rare thing.
I am happy to have a DC distribution network in my house, with the number of devices that use DC voltage in my house it would not be easy for them all to have their power supply connected to the wall socket.

J. Vieira

yozik04, 2024-07-25:
@jmcvieira1's reminder led me to inspect the board's power connections, suspecting the header pins might be a weak link. Upon reopening the box under better lighting, I was greeted by the sight of an eight-legged deceased extraterrestrial. 😄

[image]

Turns out, the real culprit could be the conductive spider web, and possibly its former occupant. If that's the root of our issues, my apologies for the wild goose chase we've been on.

yozik04, 2024-07-25:
Also an update from
```
Link Error Statistics
31 0000009896
32 0000001562
33 0000860000
34 0000000000
35 0000000000
```

> The byte that says "86": "8" indicates a Stack Overflow occurred at some time.

So there was a Stack Overflow recently.

I will clean up stats one more time after clearing spider body and it's web.

jmcvieira1, 2024-07-25:
I hope you found the culprit, if I were you I would remove the two Philips screws and see if you don't have the rest of the spider's cocoon under the PCB, with a little moisture it is an excellent electrical conductor, and will cause a lot of problems without any sense.

J. Vieira

nielsonm236, 2024-07-25:
This is a new one on me. While everything is conductive at sufficient voltage, I've never heard that spider webs we sufficiently conductive to cause a problem like this. Maybe you are playing a joke on me?

With regard to the Stack Overflow - yes that is a problem. I will look into it.
On looking back through our conversation I see Jose's module also shows a Stack Overflow.

I had false reporting of Stack Overflow in the Upgradeable builds for awhile. That was cause by the Code Uploader taking advantage of memory that crossed the Stack limit. That was OK because the Stack was not using the memory in that specific scenario. I fixed the error in that case. But it should never happen in the non-upgradeable builds.

Mike

jmcvieira1, 2024-07-25:
No tricks, I've already repaired some sliding gate controllers just by cleaning the cobwebs and especially the cocoons, if they are made in the right place, on a humid day the gate even opens and closes by itself, it seems like it took on a life of its own ...
From the photo, the spider is there close to the crystal oscillator, it doesn't take much to disturb the functioning of the processor, and we haven't yet seen underneath the board, where it's warmer, it's the ideal place to make the cocoon.

With regard to the Stack Overflow my Link Error Statistics are from the 20231009 1022 MQTT Domo UPG .

J. Vieira

nielsonm236, 2024-07-25:
I learn something new every day.
It's possible the Stack Overflow in your setup occurred due to the Code Uploader problem. I released the fix in the 20240612 build (just a month ago), although I actually fixed the bug back in December in the development code. But please keep an eye on that. It is possible there is a Stack Overflow problem in the base code like @yozik04 is running that also appears in the base code of upgradeable builds. At least this is a little easier to track down. I will get on it.
Mike

jmcvieira1, 2024-07-25:
I just updated to the latest firmware.

Code Revision 20240612 0226 MQTT Domo UPG .

Link Error Statistics
31 0000000105
32 0000000054
33 0000860000
34 0000000100
35 0000000000

J. Vieira

nielsonm236, 2024-07-25:
Did you run /67 to clear the old error?Sent from my iPhone

jmcvieira1, 2024-07-25:
No I didn't, I'm sorry, rookie mistake, I already edited my post after run the /67
the 8 is gone for now.
Link Error Statistics
31 0000000006
32 0000000038
33 0000060000
34 0000000000
35 0000000000
J. Vieira
Thank you

nielsonm236, 2024-07-26:
I've been running a test for the past 24 hours with Code Revision 20240612 0226 MQTT Home. The test uses HA to turn on outputs 1, 3,4,5, 6 every 5 minutes, then off 30 seconds later. I don't have DS18B20's attached. So far I'm not seeing any errors (not even TX or RX errors).
Next I will attach 4 DS18B20's, and I will set up some HA automation on the 8 other modules in the test bed to attempt to generate a large amount of HA and Broker activity in hopes of provoking some ethernet collisions. Will let you know after that runs at least 24 hours.

tomaszvil, 2024-07-27:
Hello Mike Hello all
I'm following your issue comments
I noticed that colleague  @yozik04  has a similar problem to mine so I decided to return to the discussion.
I want to share my observations:
I took the same module with the same power supply that was losing communication from the workshop to my home where the network works without large flows of information 2-3 computers 2-3 smartphones. It has been working stably without anomalies for four months.
When I had the the same module in the workshop where I have over twenty computers in the network, several NASs, several spartphones, over twenty IP cameras, every few days I had a complete loss of communication.
I checked with an oscilloscope the interference on the power supply of the modules was lowly and negligible. For few days  I even had a laboratory power supply connected to the HW584 and the problem also occurred.

That's why I will insist that the problem is not in the power supply but in the information that circulates in the network and is incorrectly interpreted by HW-584.
Often, the information that is stored in RAM is damaged because there is communication with the module but the displayed MAC is incorrect, then it loses communication with HA (after resetting the module HW584 restores the correct MAC).
It happened (rarely) that the content in the processor EEPROM is damaged, also the MAC data saved in the EEPROM cells, I read them several times and analyzed the EEPROM. Sometimes other areas also changed in EEPROM.

I will only add that I recently changed my router from CISCO busines to UBIQUITI, the problem still occurs...
sorry for my english

greetings to all Tomasz

nielsonm236, 2024-07-27:
Hello Tomasz - That is an interesting observation and aligns with my findings that the ENC28J60 allows traffic to pass through, that requires firmware on the STM8 to do the filtering. The ENC28J60 is supposed to only allow traffic that matches the MAC address, or that matches broadcast requirements, to get through. What I observed is that there are times when a very large amount of traffic is "broadcast", and I found that (on my network) some of it appears to come through the firewall from the Internet, some from Servers, and some from the now ubiquitous Smart Home devices. I imagine there are many other sources as well, including firewall and network security software.

I have 17 IP cameras on my network, but they are all polled from a server, so I don't think the cameras themselves add to the broadcast traffic in my case. It is likely that most IP cameras broadcast their presence so that they can be found by media devices. I have 22 computers (all Windows, 1 sometimes booted to Linux), but these are for testing and usually only 3 or 4 are powered up. Typically there are only 3 smartphones on the network unless there are guests. So, while I can see a lot of broadcast activity on my network, it is probably nothing next to networks with many more servers and devices.

SO, if we assume that network traffic, likely broadcast traffic, is a problem (as your test seems to strongly support), there are two ways it can bring the HW-584 down:
1) The ENC28J60 hardware buffer over-flows. This should just cause dropped messages which need to be retried. But at high enough levels of traffic overflow it becomes a self multiplier: Dropped messages require more resources and more retransmits, further exacerbating the hardware buffer overflow.
2) If unnecessary ethernet traffic requires the STM8 firmware to process and discard all the traffic ... well ... that is a very low horsepower device and I can easily see it falling behind on the traffic.

Can we improve the situation? If I can find a way to reliably generate the necessary network traffic I might be able to tweak some aspects of the firmware to more quickly clear the ENC28J60 hardware buffer. As it is all I can do is perform some more code inspection on an assumption that I need to discard certain traffic more quickly. AND I should also be looking for a bug in the RAM buffer pointers that might be occurring because of very high levels of traffic. But, I must admit that improved filtering only sets a higher threshold before failure. 

We have observed similar problems occurring because of device resets caused by the power source, I suspect because the device gets several partial initializations interspersed with unexpected power loss. But I'm also convinced you did a very thorough job of verifying that power is not the problem in your case, and network traffic is the most likely source.

(Note to self: Maybe the filter could be improved by a) ignoring all broadcast traffic at the start of the filter, as I don't think the device utilizes any broadcast except for Ping Echo Requests, and even that might not be necessary, b) allowing traffic only from a specified IP range, but I suspect that filter can only be applied after we are well into over-run. Not sure any of this would be adequate for high broadcast networks, and in the end we are limited by STM8 horsepower).

Mike

nielsonm236, 2024-07-27:
@tomaszvil and @yozik04 Here is something you could try if you suspect network activity is the source of the problem. If you have a module you can connect to the network and you load it with one of the Browser Only version (either SWIM or UPG based) you can use URL command /68 to get what I call a "Network Statistics" report. Here is a screenshot of a Browser Only module on my network. From the /66 command I know this module has been online 257 hours since I last rebooted. Look at how many packets were "Dropped at the IP layer" because they did not apply to this module. And my network is apparently quieter than yours.
[image]

@tomaszvil You could try this on your home network to get a baseline number, then try in in your workshop to see what difference there is. It is just a way to get more clues.

I'm going to look at the uip code to see if I can add a count for "Dropped Broadcast Packet". That would at least confirm for me if most of the garbage traffic is Broadcast originated.

Mike

nielsonm236, 2024-07-27:
@tomaszvil Maybe wait until I look for additional data collection that can be done in the Statistics.
Knowing that the module worked fine at home but not at work was a big reveal to me.
Mike

nielsonm236, 2024-07-28:
Here's some test code and more information on what the Statistics tell you.
[20240728 Browser Test Version.zip](https://github.com/user-attachments/files/16402795/20240728.Browser.Test.Version.zip)

First, the front end of all HW-584 modules are seeing the same network traffic. So running a module with the Browser firmware will tell you what traffic all the modules are seeing (with one exception: traffic sent to a given module MAC with the wrong IP Destination Address).

CORRECTION to something I said earlier. If you look at Link Error Statistics (URL /66) and you see a count in the lowest two characters of Line 33, that is the ENC28J60 RXERIF error count. That indicates that the Receive Hardware Buffer in the ENC2J60 has been over-run with network traffic and packets are being lost. It's OK for that to happen infrequently, but likely prevents operation if it happens more than once every few minutes. Of course, you prefer that it never happens.

The attached firmware includes an improved Network Statistics Display. The biggest change is that it now shows detail on packets dropped because they are a) Broadcast packets, or b) Packets that were sent to the module MAC, but did not have the correct IP Address. The new statistics confirm what I said before: Even on my relatively quiet network there is a very large amount of Broadcast packet activity. Here is a screen shot after running 1 hour:
[image]
Note the number of dropped packets, and that they were almost entirely Broadcast packets. It is only about 4 packets per second (which is really nothing),
Now consider that the Receive Buffer in the ENC28J60 is 6KB, and that Broadcast packets can be anything from about 40 bytes to over 65000 bytes. ANY packet over 6000 bytes will cause a hardware over-run and will be discarded. We don't care about the discard, but we DO care about preventing other packets from being received while a useless packet occupies all the resource. At least on my network no large packets are being seen.
Also consider that in a very busy network the number of Broadcast packets could be much, much larger. So even if they are small packets, a large number could fill the buffer and interfere with "good" traffic.

Doing a little online reading about excessive Broadcast packets there is mention of a number of sources that could cause a Broadcast storm. Usually misconfigured equipment. Typically not a problem, but could be.

@tomaszvil You mentioned you have IP cameras. I know IP cameras can be set up to Broadcast, and of course video packets can be quite large. So that is worth looking into if Broadcast really isn't needed. I have my IP camera host poll the cameras rather than have the cameras broadcast.

That's an update. Armed with the statistics we might be able to figure out if Network traffic is the issue, and perhaps what the source of the traffic is. If it is determined that all the Broadcast traffic is essential, then the HW-584 simply won't work on that network. Perhaps using a sub-net might then be a solution to isolate the HW-584 from the Broadcasts.

Mike

nielsonm236, 2024-07-28:
Quick update: I may have found a bug. The ENC28J60 spec with regard to it's front end filter is a bit confusing. But it looks like I can have the hardware reject Broadcast packets before they ever get to the firmware. I will start a test on that now and send new firmware if successful.

yozik04, 2024-07-28:
There are no new updates to report; everything is functioning as smoothly as it has been. There have been no transmission/reception errors or stack overflows. I'm considering applying a plastic spray that I have, which is intended for electronics, to coat the board with a dielectric layer to protect it.

nielsonm236, 2024-07-28:
Did it only require removing the spider webs? Still using the same power supply?

yozik04, 2024-07-28:
I only removed spider webs and disabled full duplex. Same power supply, same everything.

yozik04, 2024-07-28:
You need new test bench, with spiders =)

nielsonm236, 2024-07-28:
@tomaszvil After applying the ENC28J60 hardware filter to reject all Broadcast packets, here is the Network Statistics report after 9 hours of runtime. Now the firmware see's no Broadcast packets. I am not sure if Broadcast packets could still overwhelm the ENC28J60 buffering system, but certainly the hardware will reject the packets and keep the buffer cleaner than the firmware can.
[image]
I will prepare Test code for the Browser and MQTT HA versions and distribute within this Issue shortly.

nielsonm236, 2024-07-28:
@yozik04 I always knew I couldn't try every scenario in test. But I NEVER thought spiders would be a required test :-)
Because of the location of the spider web I'm going to guess it was interfering with the oscillator somehow. This one goes in the "Believe It or Not" category.
Because you can hear noise in the power supply I would remain cautious about it. The cheap switching supplies often make noise (or smells or smoke) as they pass away.
Tomaszvil's evidence points to network traffic very strongly as the source of his issue. Very odd that the symptoms are similar. But it is possible that more than one cause can create the same symptom, IF all those causes interfere with traffic processing, and that then creates the symptom we see.

nielsonm236, 2024-07-28:
Here are test builds of code that includes hardware filtering to eliminate Broadcast packets from ever making it to firmware. In very busy networks this should improve performance of the HW-584.
@tomaszvil I'm hopeful you could give this a try in your workshop network.

[20240728 Broadcast Filter Test Version.zip](https://github.com/user-attachments/files/16405219/20240728.Broadcast.Filter.Test.Version.zip)

Mike

tomaszvil, 2024-07-28:
Hello Mike
You work so fast that I can't keep up with reading...
but I'm already testing 20240728 Broadcast Filter Test Version (I chose UPG version)
Does 20240728 Broadcast Filter Test Version have firmware includes an improved Network Statistics Display URL /68?
Tom

nielsonm236, 2024-07-28:
Yes, the Test code has the updated Network Statistics. If it seems to be helping at all then letting be know the Link Statistics and the Network Statistics.
Remember: Network Statistics only works in a Browser Only build. So we need to use one of the Browser builds (SWIM or UPG) to get a feel for what is happening in the Network. If that looks clean, then you can load the MQTT build I sent. It also has the Broadcast filter, but does not have the Network Statistics report feature.
I have my fingers crossed that we are onto the problem.
Mike

nielsonm236, 2024-07-28:
@jmcvieira1 Jose - I included a Domo build with the Broadcast filter in case you want to try it. But I think your network is like mine .... not busy enough to have had a problem anyway.

nielsonm236, 2024-07-29:
Caution - After running many hours the module running the new Browser UPG build stopped running. I need to look into this.
Mike

nielsonm236, 2024-07-29:
Well, this felt very close to vistory, but there is some kind of problem. Perhaps completely blocking Broadcast messages is not a good idea. As I looked into this I found that when a module became unresponsive I could restore operation by loading the CodeUploader (in which I inadvertently still let Broadcast messages through) and the module would then respond, Then I could load firmware that was again blocking Broadcast messages ... and it would start responding again ... but I don't know for how long. I need to study this further.

[94 more replies]