nielsonm236/NetMod-ServerApp #225: Feature Request:  Web Version:  Get Temerature Data command

guytpetj, 2024-06-26:
Is it possible to add to the browser only version a command to get the temperature sensor information (sensor ID, deg C, degF for each sensor) in the same way as commands 98 and 99 work? I'm in need for this for a swimming pool controller replacement project.

nielsonm236, 2024-06-26:
Yes ... but let's discuss.
Remember that any Browser command returns it's answer in the form of a webpage. The webpage response can be more complex, like the existing IOControl and Configuration webpages. Or the webpage can be simpler, like the response to /98 /99. Regardless you need a mechanism to decompose the webpage response into the data you want.
Since the above is a fact, and I know you have to write a mechanism to extract the sensor data from whatever page is received, an answer could be "just extract it from the IOControl page you already get with URL command /60".
However, recognizing that maybe you want the extraction process to be simpler, then "Yes, a special page could be generated that is simpler in design". Question:
- Does the data extraction software you plan to use have requirements for defining the data fields?
- Would a page that simply provides fields that look like this:  3c01e076be0f,027.8,082.1 be sufficient. (Note commas for field separators). The order would always be ID, DegC, DefF. The temperature fields are always 5 characters. The leading 0 would be replaced with a "-" character for negative temperatures. Of course, if this is a swimming pool I hope negative never happens, but allowing it makes this more generally useful.
- If the above is not adequate, do you have a suggestion?
Mike

guytpetj, 2024-06-26:
Hi Michael,

I use Raspberry Sharp (C# lite version) to extract the data. This is limiting what I can do as not all C# commands are available, and I can’t include additional libraries.

The following is the code I use to get the data used to set the software switches as per the switch status of the Web_Relay_Con PCB.
I can uses similar code to extract the temp data if it was available in the same way as /98 or /99.

    WebRequest request = WebRequest.Create(http://10.1.1.7/99");
    WebResponse webresponse = request.GetResponse();
    Stream dataStream = webresponse.GetResponseStream();
    StreamReader reader = new StreamReader(dataStream);
    NetStatus = reader.ReadToEnd();

This is exactly what I’m looking for. When I have the data in a string, I can do whatever I want.

  *   Would a page that simply provides fields that look like this: 3c01e076be0f,027.8,082.1 be sufficient. (Note commas for field separators). The order would always be ID, DegC, DefF. The temperature fields are always 5 characters. The leading 0 would be replaced with a "-" character for negative temperatures. Of course, if this is a swimming pool I hope negative never happens, but allowing it makes this more generally useful.
Negative values do happen in the winter where I use the ambient temperature sensor to switch the pool to low speed circulation mode to prevent the equipment from freezing. So having negative temperature is great to have. I use 5 temperature sensors for the pool control, so that fits perfect with your code.
Regards,
Peter

guytpetj, 2024-06-27:
PS: I forgot to mention that the Fahrenheit temperature here (Houston, Texas, USA) often goes over 104 degrees in the summer, so got to make sure that 0104.0 is possible.

Regards, 
Peter

guytpetj, 2024-06-27:
> Example temperatures would look like this so that they are always 5 characters:
> -12.1
> 000.0
> 051.5
> 104.6
> Your data extraction only needs to check if the first character is a "-" or a digit to know positive or negative.
>
> Five sensor fields would be reported on the webpage regardless of how many sensors are present. For the moment I'll assume it is OK to have a newline between sensors (unless you tell me otherwise). So it would look like this (a case for 4 sensors):

Hi Mike,

That looks good. My preference would be a single line with all data comma delimited, but it is easy to deal with multiple lines as this may improve readability for other users.

I hope you can make this happen in one of the next releases. For sure I will appreciate it a lot.

Currently I use the web relay  controller in conjunction with a Raspberry Pi and HomeGenie in which I programmed a fully functioning,  scalable, and IOT Pool Controller, but the RPi is struggling with the extreme summer heat we have, the web relay controller performs fine in this heat. The only thing currently stopping me from separating the RPi and controller, placing the RPi inside the house where it is cool, is the need for 5 x DS18B20 temperature sensors. This is where your firmware is the answer to my problems.

Thanks,
Peter

3c01e076be1c,027.8,082.1
3c01e076be2d,027.8,082.1
3c01e076beff,027.8,082.1
000000000000,000.00,000.0

This could also be one long string of characters with a comma (instead of a newline) between sensors. Might be slightly harder to debug as that is less human readable, but not a big deal. Let me know what you think. Also, doesn't need to be a comma delimiter ... could be a space if you think that is better.

Note: I've not committed to do this just yet, but at face value it looks like a fairly simple request. It always seems to happen that "simple" becomes "problematic" when I start actual code work, but so far I'm not concerned. Fortunately the "Browser Only" versions have some extra Flash space available.

nielsonm236, 2024-07-01:
I'm working on the code, and to minimize code size I'll have to change the spec a bit. I'll let you know what we need to do shortly.

guytpetj, 2024-07-02:
Just as a thought, why not add it to page 98 or 99 as:
0000000000000000,3c01e076be0f,027.8,082.1,3c01e076be0f,027.8,082.1,3c01e076be0f,027.8,082.1,3c01e076be0f,027.8,082.1,3c01e076be0f,027.8,082.1

or ignoring the Fahrenheit and only show Sensor ID and Centigrade as this is easily converted in software:
0000000000000000,3c01e076be0f,027.8,3c01e076be0f,027.8,3c01e076be0f,027.8,3c01e076be0f,027.8,3c01e076be0f,027.8

or ignoring the Fahrenheit and sensor ID, only show Centigrade as this is easily converted in software, and the sensor ID's are known from page /60 for proper assignment of the temperature.
0000000000000000,027.8,027.8,027.8,027.8027.8 ( Relay State, Temp1 C , Temp2 C, Temp3 C, Temp4 C, Temp5 C )

Like I said, just a thought.

nielsonm236, 2024-07-02:
I'm almost finished with a test code build, but have a question. I had forgotten the details of the temperature conversion and display code, so I had to do a little review. The result of the code addition looks like this:
253e656c7469, 027.4, 081.3,000000000000, 000.0, 000.0,000000000000, 000.0, 000.0,000000000000, 000.0, 000.0,000000000000, 000.0, 000.0
On this test board I only have one sensor, but the above illustrates the output. There is no HTML formatting, so what you receive in the payload will look just like the above.
The question: The temperature values are always 6 characters (not 5 as I described earlier). Included in the 6 characters: Positive temperatures are preceded by a space. Negative temperatures are preceded by a minus sign. Does this work OK? Or would it be better if positive temperatures are preceded by a '+' instead of a space?
After your answer I should be able to post a TEST build.
Mike

nielsonm236, 2024-07-02:
I just noticed your earlier comment.
I don't want to change the /98 /99 output as some people already use it. I'm using /97 for the new temperature page.
I suppose we could just send temperatures in C. I can test the build sizes to see if it helps much.
Mike

guytpetj, 2024-07-02:
Thanks Mike, this is looking good.
It may be batter to do a + than a space, it makes it more obvious for others.

Regards,
Peter

nielsonm236, 2024-07-02:
You can give this a try. I've only tested with the UPG version, but it should work fine with the standard SWIM version.
This is a TEST load only. I'll get it into a release as soon as I can.
Mike
[20240702a Test Code.zip](https://github.com/user-attachments/files/16073771/20240702a.Test.Code.zip)

nielsonm236, 2024-07-02:
Hmmm ... might have a bug here. I just hooked up 4 sensors for more testing and what I'm getting in the short form display doesn't match what I see in the IOControl display. I'll have to figure out how this is getting scrambled.
Mike

guytpetj, 2024-07-02:
I just wanted to report the same, the information of /60 and /97 do not match. Sensor ID and Temperatures are wrong in /97

Regards,
Peter

nielsonm236, 2024-07-02:
OK - Fixed it.
[20240702b Test Code.zip](https://github.com/user-attachments/files/16074837/20240702b.Test.Code.zip)

guytpetj, 2024-07-04:
Hi, I've run the test version for 24 hours with 5 sensors attached using the standard browser version. So far everything works as per expectations.

Thanks, 
Peter

nielsonm236, 2024-07-04:
Excellent. You can continue your development with the test code, and I will release it with the next general release. That might be a couple of weeks out.