nielsonm236/NetMod-ServerApp #228: Last will and availability after reconnect

yozik04, 2024-07-25:
While we were investigating #221 I've noticed another problem

MQTT Explorer started to show `availability=offline` after working properly for 27 hours. `availability=offline` happens when MQTT device looses connection to Broker and Broker executes `last_will`. Does the unit send `last_will` again with retain flag if it reconnects?

nielsonm236, 2024-07-25:
I will check. But I think last_will is sent only at boot time with retain.

nielsonm236, 2024-07-25:
From code inspection:
If a disconnect occurs the device will restart the MQTT connection. When the device reconnects it will resend the Last Will with Retain set.
The disconnect can be sourced from either the Broker or from the device.
The device will generate a disconnect when MQTT error counts exceed certain limits or when a Ping Timeout occurs.
Ping is set at 60 seconds.

yozik04, 2024-07-25:
So the Last Will should be sent after a disconnect event sourced from the Broker as well?

nielsonm236, 2024-07-25:
As FYI, when a reconnect is performed I rotate through MQTT port addresses. I'd have to go look up why I did that, but attempting to use the same port causes some timeout problem at the Broker lasting 10;'s of seconds. Using a new port gets around that timeout problem.
Mike

yozik04, 2024-07-25:
Thank you for explaining. Will close for now.

closed 2024-07-25 by yozik04

nielsonm236, 2024-07-25:
Issue #159 (closed) describes why a port rotation is required. It's not the Broker ... it is Windows that makes this necessary.