nielsonm236/NetMod-ServerApp #234: 16 Relay Module compatibility with Network Modules V2

joaoanlopes, 2024-09-10:
I m working on a project were I'm using 16 Network Modules V1 controlling 16 units of '16 Relay Module' boards.
Since I faced the problems on using this version of the Network modules I decided to substitute all of them with the V2 version ( Hope to get them next week via Ali Express) and then flash them with this corrected firmware.
I allready read the installation manual created by Nielson and I hope I get all the directions I need. 
Anyway I steel can't get If I need to substitute or modify all my 16 Relay Modules to work with the new HW-584. As long as I understand some versions need to be modifyed but there another ones that will work directly with it.
Since I'm a beginner on eletronics its a little confusing to me to understand all the steps to make my relay boards working with the new modules arriving and I'm considering to substitute the too with the model sugested on the manual 

( https://pt.aliexpress.com/item/1005004962029461.html?spm=a2g0o.cart.0.0.79687f06E9vIvm&mp=1&gatewayAdapt=glo2bra )

Please check some photos of my setup and the boards Im talking about.
I will appreciate all your opinions and advises about the better decision to make on this. Thanks in advance Mr Nielson

**- 16 Relay Module version I m using »**

[image]

[image]

[image]

**- Complete setup »**

[image]

nielsonm236, 2024-09-13:
Impressive setup. That's a lot of work!

The 16 channel relay board you identified at this link
( https://pt.aliexpress.com/item/1005004962029461.html?spm=a2g0o.cart.0.0.79687f06E9vIvm&mp=1&gatewayAdapt=glo2bra )
does look like the one that contributor P-O Yiniemi (peoyli) identified in Issue #202 as being IO signal level compatible. I never bought any so I can't speak to it in detail. And if you buy them you should probably buy just one at first to confirm it works properly.

If you decide to modify your existing boards you'll probably need to use Fix 1 or Fix 2 from the manual. In either case if you want the 5V regulator on the Relay board to provide power to the HW-584 (the V2 board) you will need to make sure the 5V pins on the connector to the HW-584 actually have 5V on them (making sure the Fix applied only reduces the voltage to the resistor packs and NOT to the HW-584 connector). Fix 1 leaves the 5V regulator intact, Fix 2 replaces the regulator which will require a separate power supply for the HW-584). So, having said all that Fix 1 is likely to be preferred for you.

All this is difficult to describe in words. I'm hopeful the manual does an adequate job.

Mike

Stefan4691, 2025-01-08:
> limitations on V1 for example all of them using the same MAC addresses. It really drives me nuts.

I operate the little brother of your project and also faced the 'same MAC' issue.
If your switch and your Router supports VLANs, you can assign any single V1 board to a separate VLAN, thus any V1 would require it's own IP address.... in this case the duplicated MAC addresses don't hurt in my case. 
Do not pay too much for the switch, as some more expensive ones will complain about the 'same' system appearing in different VLANs