nielsonm236/NetMod-ServerApp #243: Unable to install the UPG firmware ["???" error on the webpage]

geoprea1989, 2024-11-20:
Hello,

After struggling for two days I finally decided to ask for help :)

I've been using the HW-584 modules with the normal version with MQTT and and Home Assistant [non-UPG] provided by Nielson for over a year, but since the beginning I was having issues with the webpage becoming not reachable on all the 4 modules I installed in different places, with the only way to regain the access to the webpage being the restart of the module or wait hoping that it will recover randomly.

After reading the wonderful user guide I seen that I can do more with it, most important to add the PCF8574 module as I needed to expand the number of I/O pins and I decided to buy from Aliexpress two 24LC1025 EEPROMs and go for the UPG version.

I found the SMD versions, but I had the DIP PCB adapters already available from past projects.

Also, I had the blue boards with the socket in my project box, but I bought some spares, just in case :)

After programming the STM board with the SWIM interface [NetworkModule-CodeUploader.sx] I connected the EEPROM board to it and give it a try, but unfortunately I was facing the "I2C EEPROM Missing" error on the webpage :(

I tried the spare 24LC1025 EEPROM and I faced the same issue.

I then decided to move the A2 jumper back to default [tied to GND] instead of VCC as in the user guide and the EEPROM was visible by the STM board and the Code Uploader webpage loaded successfully.

Given so, I went forward by first uploading the "NetworkModule-strings.sx" file and afterwards the "NetworkModule-MQTT-Home-UPG.sx" one, but I faced another problem, the "�����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������" message.

I think it's the one TurboThronix faced in his post titled "getting a bunch of quetionmarks #97".

I tried clearing the flash with /72 and the EEPROM with /72 and doing the same process again for a few more times, trying other versions, like Domo, BME or even the Browser versions, all of them being UPG, but in the end I was always facing the "???" message :(

I cleared the STM flash [both program and data memory] by toggling the "Read Out Protection" bit, just in case there was a glitch, and rewriting the "NetworkModule-CodeUploader.sx" file, but in the end I was still getting the "???" message.

I was thinking that maybe I got scammed and received face 24LC1025 modules, but I connected them to my trusty CH341A programmer and read the whole memory to check for inconsistency.
After that I changed the last bytes, performed a full rewrite and rechecked if they're still there instead of the FFs and it seems that the EEPROMs are legit, they hold the data in the last bytes.

As a test I filled the whole memory of the EEPROMs with data and it matched it too, so the integrity of the data was confirmed as well.

Going back to the A2 pin from the module, I decided to try a few EEPROMs that I had in my project box, and it seems that even with other EEPROMs like the 24C256N, if the A2 pin is tied to VCC, the STM is unable to read it and triggers the "I2C EEPROM Missing" error on the webpage.

I replaced the blue module with the DIP socket with another ones from my old "stock", but still the same.

I tried downgrading everything and trying all the files from the previous release "20231009 1022 General Release" but I'm still getting the "???".

The same behavior was faced from all the browsers I checked: Firefox, Chrome, Edge, Edge with IE Mode enabled [Compatibility Mode enabled].

In the end, I reflashed the "classic" MQTT file ["NetworkModule-MQTT-Home.sx"] to check the sanity of the STM board and it fired up the webpage I was used to on the first restart, yeeey! :))

So, I'd really love to give it a try to the UPG version, but I just don't know how to get past the "???" error I'm facing and I'd really appreciate your help on this.

Also, maybe we can find why the webpage becomes unresponsive on the "NetworkModule-MQTT-Home.sx" version and it recovers randomly for a few minutes and then it goes rogue once again and repeats the cycle.
BTW, once the webpage becomes unavailable, the Home Assistant sees it offline too, and the board only replies to ICMP [ping] packets as usual, with no different latency than normal [I was thinking that it hangs].

Sorry for my really long post, but I was thinking that it will ease your troubleshoot if I tell you all the things I done from my part. I hope I haven't forgotten something, haha :D

Thank you so much for your help and, last but not least, many thanks for developing and maintaining this project, I know a lot of people relies on it :)

Thanks!

nielsonm236, 2024-11-21:
I apologize for my delay in responding. A bit busy with family issues. Seems retirement doesn't leave you with as much time as you might think :-)

Your opening comment " ... I was having issues with the webpage becoming not reachable on all the 4 modules ... " has me concerned, but since the originator of issue #221 has gone silent it might be an opportunity for me to explore the source of that problem with you. 

But first let's go on to your problems installing the UPG code. The issue you are seeing is usually because the "NetworkModule-Strings.sx" file was not uploaded to the module. Obviously you got the Code Uploader installed. But then you need to upload NetworkModule-Strings.sx, followed by the whatever UPG code you want to run.

As FYI, the NetworkModule-Strings.sx file contains the webpages used by the UPG code. They are stored in the I2C EEPROM to free up space in the STM8 Flash memory to allow more code features.

Let me know if that helps.

Mike

nielsonm236, 2025-01-30:
I'm going to assume my comment resolved this issue.

closed 2025-01-30 by nielsonm236