nielsonm236/NetMod-ServerApp #244: FW programming did not work: RESET LED red after FW install...

Stefan4691, 2025-01-07:
What a great project! Thank you for the excellent work and detailed documentation. 
Everything looked very straightforward.

------------
This is a brief introduction to my project. You can skip this section and jump to the next block if preferred.

For the past 7 years, I’ve been running six 16-channel relay cards in my house to control lights, door openers, and heating valves. These are v1 boards, which have worked fine so far. I overcame the MAC address issue by configuring VLANs, each with only one device. Luckily, my boards retained their IP addresses.

However, for the past few months, the boards have started to hang and become unresponsive. To address this, I began developing a similar solution using ESP32 boards, but I struggled with the 3.3V compatibility issue (which you described in detail). While searching for a solution, I came across your project. Since I use REST-style calls with my current setup, I decided to adopt the v2 boards and your firmware. It seemed like the most elegant way to keep the heating system running with minimal development effort, requiring only the configuration of links to the relays. Your web pages are divided into 2x8, while my boards are arranged as 4x4, so I made minor adaptations to reduce the number of 'next page' links required. I ordered eight v2 boards to implement this solution.

I encountered the same 3.3V compatibility issue with the relay boards. Since my boards are fully integrated and wired, I didn’t want to uninstall them. After studying the design, I decided to remove the B170 MOSFETs using a wire cutter (I know... it hurt a bit). I then soldered male-ended jumper cables to the drain contacts, connecting them to the Darlington arrays. After making these adjustments, I tested with the original firmware, and everything seemed to work fine.

As I use Home Assistant, I wanted to utilize your firmware with the REST implementation and eventually transition to the MQTT configuration.

-------------------
I followed all the instructions in your documentation. However, I didn’t follow your recommended path as I don’t plan to modify your code. I only want to use the NetworkModule-MQTT-Home.sx firmware.
I am using General Release 20240612 0226, and everything you promised happened in the STVP:
I received the out-of-range messages, successfully programmed the ROP, and successfully programmed the firmware.

PROGRAM MEMORY status:
[0x008000 - 0x00FFFF]
File: NetworkModule-MQTT-Home.sx
Checksum: 0x2C2D19
Programmed
Memory checksum: 0x2C2D19

However, the board is not starting. The reset LED stays red, and the green Ethernet LED shows network activity.

Do you have any idea what I might be missing?

Thanks a lot.

nielsonm236, 2025-01-08:
If the Red LED stays on that is expected ... so it is possible the firmware is actually running and there is some other issue. A couple of suggestions:
1) Try cycling power to the board. When it starts you should see the LED flicker a few times as the boot processes run, then the LED will stay ON.
2) If #1 seems to be working, then we need to figure out why you can't access with your Browser. The first thing that occurs to me is that the default settings for a freshly programmed module are as follows:
Device Name			NewDevice000
IP Address			192.168.1.4
Gateway Address		192.168.1.1
Netmask			255.255.255.0
HTTP Port			8080
Features			All boxes unchecked
MAC Address			c2 4d 69 6b 65 00
MQTT Server IP Address	0.0.0.0
MQTT Port			1883
MQTT Username		NULL
MQTT Password		NULL

Note that the HTTP Port is 8080. So, when you first attempt to contact the module use URL 192.168.1.4:8080. Send me a screenshot of what you see with your Browser.

If you Network is not based on 192.168.1.xxx IP addresses let me know and I can point you to how to modify the default settings via the a couple of different methods.

Mike

Stefan4691, 2025-01-08:
Hi Michael,

you are right. it seams to work. The only change I made was to power it off for a day.

I was first using the original FW and accessed it at 192.168.1.4. The LED was always off.
I updated it and could not attach to it anymore and the LED is permanently ON now.
Even power cycling it or connecting it p2p to my Laptop did not help.

But as mentioned, I tried it again after your message, and must observe the stability, now. No clue, what happened.

I really thank you for your hints.

nielsonm236, 2025-01-09:
Good that you are underway!
I will close this issue, but please open a new one if there is another problem.

closed 2025-01-09 by nielsonm236